carla-simulator/carla #5329: Bounding Boxes are too large

aeon0, 2022-04-06:
CARLA version: 0.9.13
Platform/OS: win10
Problem you have experienced: Boundingboxes are too large
What you expected to happen: Boundingboxes to fit tighter to the vehicle
Steps to reproduce: Show any boundingboxes in your vis (e.g. from examples run client_bounding_boxes.py)
Other information (documentation you consulted, workarounds you tried): The root seems to be within https://carla.readthedocs.io/en/latest/python_api/#carla.Vehicle bounding_box

Carla Box:
[image]

I was hoping for sth closer to the green box:
[image]

Any way to improve on this?

MarcelPiNacy-CVC, 2024-05-17:
Hi,

Does 0.9.15 solve your problem?

Marcel

aeon0, 2024-05-19:
> Hi,
> 
> Does 0.9.15 solve your problem?
> 
> Marcel

Hi, if I remember correctly it was a human vision/perception error. I checked the pointcloud and boundingbox in 3D and it fits. The image projection also is fine.
So in the end its just a misinterpretatiom from how I perceive 2D to 3D.

closed 2024-05-19 by aeon0