carla-simulator/carla #5330: Bounding boxes for parked vehicles

makaveli10, 2022-04-06:
Based on the [closed issue](https://github.com/carla-simulator/carla/issues/2343) I tried to draw bounding boxes on parked cars. As mentioned in the issue that bb for parked cars are in world coordinates so, I assumed that we need to convert those to sensor coordinates which I did(replaced _vehicle_to_sensor with _world_to_sensor) by adding the following code in [client_bounding_boxes.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/client_bounding_boxes.py) :
```
class ClientSideBoundingBoxes:
    @staticmethod
    def get_bounding_boxes_parked_vehicles(bboxes, camera):
        """
        Creates 3D bounding boxes based on carla vehicle list and camera.
        """
        bounding_boxes = [ClientSideBoundingBoxes.get_bounding_box_parked_vehicle(vehicle, camera) for vehicle in bboxes]
        # filter objects behind camera
        bounding_boxes = [bb for bb in bounding_boxes if all(bb[:, 2] > 0)]
        return bounding_boxes
    
    @staticmethod
    def get_bounding_box_parked_vehicle(bbox, camera):
        """
        Returns 3D bounding box for a vehicle based on camera view.
        """
        bb_cords = ClientSideBoundingBoxes._create_bb_points(bbox)
        cords_x_y_z = ClientSideBoundingBoxes._world_to_sensor(np.transpose(bb_cords), camera)[:3, :]
        cords_y_minus_z_x = np.concatenate([cords_x_y_z[1, :], -cords_x_y_z[2, :], cords_x_y_z[0, :]])
        bbox = np.transpose(np.dot(camera.calibration, cords_y_minus_z_x))
        camera_bbox = np.concatenate([bbox[:, 0] / bbox[:, 2], bbox[:, 1] / bbox[:, 2], bbox[:, 2]], axis=1)
        return camera_bbox

    @staticmethod
    def _create_bb_points(vehicle):
        """
        Returns 3D bounding box for a vehicle.
        """

        cords = np.zeros((8, 4))
        if isinstance(vehicle, carla.BoundingBox):
            extent = vehicle.extent
        else:
            extent = vehicle.bounding_box.extent
        cords[0, :] = np.array([extent.x, extent.y, -extent.z, 1])
        cords[1, :] = np.array([-extent.x, extent.y, -extent.z, 1])
        cords[2, :] = np.array([-extent.x, -extent.y, -extent.z, 1])
        cords[3, :] = np.array([extent.x, -extent.y, -extent.z, 1])
        cords[4, :] = np.array([extent.x, extent.y, extent.z, 1])
        cords[5, :] = np.array([-extent.x, extent.y, extent.z, 1])
        cords[6, :] = np.array([-extent.x, -extent.y, extent.z, 1])
        cords[7, :] = np.array([extent.x, -extent.y, extent.z, 1])
        return cords
```

But these doesnt draw any box on the parked vehicles instead a cluster of bounding boxes at a single location.
[image]

@stevelaclasse @germanros1987 @doterop As you guys have already closed this issue any help to get this sorted out is most welcome.
Thanks

truncs, 2022-04-22:
The extent is not in world coordinates. you can use a similar method to convert the extent to world coordinates by doing the following 

```
    def _bounding_box_to_world(bbox):
        extent = bbox.extent

        cords = np.zeros((8, 4))
        cords[0, :] = np.array([extent.x, extent.y, -extent.z, 1])
        cords[1, :] = np.array([-extent.x, extent.y, -extent.z, 1])
        cords[2, :] = np.array([-extent.x, -extent.y, -extent.z, 1])
        cords[3, :] = np.array([extent.x, -extent.y, -extent.z, 1])
        cords[4, :] = np.array([extent.x, extent.y, extent.z, 1])
        cords[5, :] = np.array([-extent.x, extent.y, extent.z, 1])
        cords[6, :] = np.array([-extent.x, -extent.y, extent.z, 1])
        cords[7, :] = np.array([extent.x, -extent.y, extent.z, 1])

        world_matrix = ClientSideBoundingBoxes.get_matrix(bbox)

        world_cords = np.dot(world_matrix, np.transpose(cords))

        return world_cords
```

TR-Monash, 2022-05-13:
Can confirm that @truncs suggestion works with one small change in `get_bounding_box_parked_vehicle(bbox, camera)`.
`cords_x_y_z = ClientSideBoundingBoxes._world_to_sensor(np.transpose(bb_cords), camera)[:3, :]` should be changed to
`cords_x_y_z = ClientSideBoundingBoxes._world_to_sensor(bb_cords, camera)[:3, :]`.

closed 2022-05-27 by makaveli10