carla-simulator/carla #5339: ImportError: No module named carla

gengtsh, 2022-04-07:
hi, 
after run make launch to run CarlaUE4 and press play button, then , run python tutorial.py
but i get the error : 
Traceback (most recent call last):
  File "tutorial.py", line 21, in <module>
    import carla
ImportError: No module named carla

ps: i change the editor preference->source code from Null Source Access to visual studio code. 

is the reason of source code why i get the output error?

gengtsh, 2022-04-08:
change to python3 , it works. but unreal engine collapse very often.

molua, 2022-11-01:
I have the same problem,
OS: windows10
Carla: 0.9.11
python: 3.7
I can see carla0.9.11 package when I input ```pip list  ``` in command， but I cannot import carla

Blyron, 2024-05-03:
To solve this issue pip3 unintsll carla
make clean
make PythonAPI
make launch

Blyron, 2024-05-03:
If this solves your issue would you be happy to close this issue?