carla-simulator/carla #5342: 403 when pulling FBX SDK

PriestOfAdanos, 2022-04-07:
```
#20 1630.8 BuildUtilsDocker.sh: Downloading FBX SDK...
#20 1630.9 --2022-04-06 15:18:56--  https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz
#20 1631.0 Resolving www.autodesk.com (www.autodesk.com)... 23.72.166.107
#20 1631.1 Connecting to www.autodesk.com (www.autodesk.com)|23.72.166.107|:443... connected.
#20 1631.2 HTTP request sent, awaiting response... 403 Forbidden
#20 1631.2 2022-04-06 15:18:56 ERROR 403: Forbidden.
```

This issue can be fixed with adding flag `--user-agent="Mozilla"` to wget

closed 2022-04-07 by PriestOfAdanos

ll7, 2022-05-13:
I have the same issue while building carla in a docker:

```shell
BuildPythonAPI.sh: Success!
BuildUtilsDocker.sh: Downloading FBX SDK...
--2022-05-13 06:26:31--  https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz
Resolving www.autodesk.com (www.autodesk.com)... 104.125.70.182
Connecting to www.autodesk.com (www.autodesk.com)|104.125.70.182|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2022-05-13 06:26:31 ERROR 403: Forbidden.

make: *** [build.utils] Error 8
Util/BuildTools/Linux.mk:149: recipe for target 'build.utils' failed
The command '/bin/sh -c cd /home/carla/ &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   cd /home/carla/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   make package &&   rm -r /home/carla/carla/Dist' returned a non-zero code: 2
```

Where should I use this flag? In my docker command or do I have to modify the `BuildUtilsDocker.sh`. I use the current dev branch of carla.

```shell
docker build -t carla -f Carla.Dockerfile .
```

max-lennon, 2022-06-10:
> I have the same issue while building carla in a docker:
>
> ```shell
> BuildPythonAPI.sh: Success!
> BuildUtilsDocker.sh: Downloading FBX SDK...
> --2022-05-13 06:26:31-- [...]

Were you able to resolve this problem? I'm having a similar difficulty.

ll7, 2022-06-14:
No, I wasn't. Sorry.

ndeshp2s, 2022-06-21:
Hi @PriestOfAdanos,

Could you please suggest how exactly "--user-agent="Mozilla" this fix needs to be applied?