carla-simulator/carla #5349: Importing Large Map Crashes Unreal Engine

aadityasp, 2022-04-08:
CARLA version:0.9.12 built from source
Platform/OS: Ubuntu 18.04

After importing Large map into Carla, When I try to Simulate, it crashes with a "Caught signal 11 Segmentation fault" error. I'm able to import a small map and everything seems to work fine, but I'm facing this issue when I try with a larger tiled map (20km x 9 km).

Attaching the crash screenshot below, Would be great if anyone can help me fix it. Thanks!
 
[image]

diespondency, 2024-08-02:
Same issue for me, I would be interested in finding a fix. My map is 10km by 10km. On the first launch of carla (after booting the computer) it works fine and I can even drive in the map, but a lot of times it has the same segmentation error.

ChengYaoJie, 2026-01-21:
I also created a 4x4km map, and I keep encountering this "Caught signal 11 Segmentation fault" error. I hope someone can help me solve this problem.