carla-simulator/carla #5355: Nothing happens when executing the example python scripts-Carla

mckvg, 2022-04-11:


mckvg, 2022-04-11:
After installing the 0.9.12 and python 3.7.9 in windows. There was dist folder including .egg and .whl, and I also added the environments variable of them, and I can enter the command through PythonAPI. However, After I installed all the requiements for the example scripts, clicked on'Play' and the scene was loaded, when I enter the examples command, Nothing happens. Who can help me and what should I do? Thanks.

[image]

[image]

[image]

[image]

mckvg, 2022-04-11:
seems that someting wrong on the clinet.get-world(). However, I do not konw how to solve it.
[image]
[image]

dikinova, 2022-07-18:
your python egg file is for 0.9.13 version.  what's the version of the carla server?

caodayue, 2022-08-10:
same problem, do you solve it?