carla-simulator/carla #5357: make import doesn't work

LucasFKobernic, 2022-04-11:
CARLA version: 0.9.13 and 0.9.12
Platform/OS: Windows 10
Problem you have experienced: I'm trying to import a custom map. After running "make import" no folder is created in CarlaUE4/content. This is happening in both versions of Carla (0.9.12 and 0.9.13). I have created the map on RoadRunner and export it using the export option for carla. 

I get the following output after running make import:

-[Setup]: [Batch params]: --boost-toolset msvc-14.2 --no-carla-materials
-[Setup]: Asynchronous jobs:  16
-[Setup]: Boost toolset:      msvc-14.2
-[Setup]: Install directory:  "D:\carla\Build\"
-[Setup]: Installing zlib...
    -[install_zlib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_zlib]: A zlib installation already exists.
    -[install_zlib]: Delete "D:\carla\Build\zlib-install\" if you want to force a rebuild.
    -[install_zlib]: Exiting...
-[Setup]: Installing libpng...
    -[install_libpng]: [Batch params]: --build-dir "D:\carla\Build\" --zlib-install-dir "D:\carla\Build\zlib-install\"
    -[install_libpng]: A libpng installation already exists.
    -[install_libpng]: Delete "D:\carla\Build\libpng-1.2.37-install\" if you want to force a rebuild.
    -[install_libpng]: Exiting...
-[Setup]: Installing rpclib...
    -[install_rpclib]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_rpclib]: A rpclib installation already exists.
    -[install_rpclib]: Delete "D:\carla\Build\rpclib-install\" if you want to force a rebuild.
    -[install_rpclib]: Exiting...
-[Setup]: Installing Google Test...
    -[install_gtest]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_gtest]: A Google Test installation already exists.
    -[install_gtest]: Delete "D:\carla\Build\gtest-install\" if you want to force a rebuild.
    -[install_gtest]: Exiting...
-[Setup]: Installing "Recast & Detour"...
    -[install_recast]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_recast]: A "Recast & Detour" installation already exists.
    -[install_recast]: Delete "D:\carla\Build\recast-0b13b0-install\" if you want to force a rebuild.
    -[install_recast]: Exiting...
-[Setup]: Installing Boost...
    -[install_boost]: [Batch params]: --build-dir "D:\carla\Build\" --toolset msvc-14.2 --version 1.72.0 -j 16
    -[install_boost]: A Boost installation already exists.
    -[install_boost]: Delete "D:\carla\Build\boost-1.72.0-install\" if you want to force a rebuild.
    -[install_boost]: Exiting...
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
-[Setup]: Installing Xercesc...
    -[install_xercesc]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_xercesc]: A xerces installation already exists.
    -[install_xercesc]: Delete "D:\carla\Build\xerces-c-3.2.3-install\" if you want to force a rebuild.
    -[install_xercesc]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Installing Sqlite3
    -[install_sqlite3]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_sqlite3]: A Sqlite3 installation already exists.
    -[install_sqlite3]: Delete "D:\carla\Build\sqlite3-install" if you want to force a rebuild.
    -[install_sqlite3]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Installing PROJ
    -[install_proj]: [Batch params]: --build-dir "D:\carla\Build\"
    -[install_proj]: A PROJ installation already exists.
    -[install_proj]: Delete "D:\carla\Build\proj-install" if you want to force a rebuild.
    -[install_proj]: Exiting...
        1 Datei(en) kopiert.
-[Setup]: Creating "CMakeLists.txt.in"...
-[Setup]:
   ###########
   # SUCCESS #
   ###########

   IMPORTANT!

   All the CARLA library dependences should be installed now.
   (You can remove all "*-src" folders in D:\carla\Build\ directory)

   You only need the ASSET PACK with all the meshes and textures.

   This script provides the assets for CARLA 0.9.12:
   You can download the assets from here:

       http://carla-assets.s3.amazonaws.com/20210730_564bcdc.tar.gz

   Unzip it in the "D:\carla\Unreal\CarlaUE4\Content\Carla\" folder.
   If you want another version, search it in D:\carla\Util\ContentVersions.txt.

-[BuildLibCarla]: [Batch params]: --server
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- Build debug:   ON
-- Build release: ON
-- Build test:    ON
-- Configuring done
-- Generating done
-- Build files have been written to: D:/carla/Build/libcarla-visualstudio
Microsoft (R)-Build-Engine, Version 16.11.2+f32259642 für .NET Framework
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

  carla_server.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server.lib
  carla_server_debug.vcxproj -> D:\carla\Build\libcarla-visualstudio\LibCarla\cmake\server\Release\carla_server_debug.lib
  -- Install configuration: "Release"
-[BuildLibCarla]: LibCarla server has been successfully installed in "D:\carla\Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies\"!
 
And then it opens Import.py

Both file .xodr and .fbx has the same name. I also tried with the tags and without it.

have anyone some idea, what i'm missing here?

Regards,

luoxinkaikevin, 2022-04-11:
I also meet the same problem 
how do you solve it ?

LucasFKobernic, 2022-04-11:
Until now i made the import manually. But I'm also facing problems with navigation file for the pedestrian. Even i have generated it, no pedestrian is being spawned.

Hanss8, 2022-04-18:
Hello,
do you have solution for this?
Thanks

LucasFKobernic, 2022-04-19:
Not yet. I made the import manually.

Hanss8, 2022-04-19:
And please can you describe me how you import your map manually in the latest version of Carla and Unreal? Thank you

luoxinkaikevin, 2022-04-23:
> Hello,
> i have same problem with make import. When i run make import the last message is:
> [13108:0420/100832.233:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is disabled, ANGLE is
>
> Is here someone to provide a solution please?
> Thanks

No. I think this problem is influenced by GPU device and the windows OS, so i try to use carla on the linux (ubuntu18) and never meet this problem.

Hanss8, 2022-04-25:
Please another problem when i run make import.
I run make import in x64 command prompt, then Visual Studio automatically opens import.py, but message Import "carla" could not be resolved is shown.
Egg file in \carla\PythonAPI\carla\dist i have.
Could you please help me?

luoxinkaikevin, 2022-04-26:
did you use the windows OS? I also meet this problem on the windows and did not resolve it.
so I have to use Linux platform to make my map, and this problem did not occured when i run "make import" command.
if you still wanna make map on windows platform, you can try the roadrunner plugin in Unreal Engine, but it will be more complexed.
Good luck.

52bloo, 2022-06-24:
> CARLA version: 0.9.13 and 0.9.12 Platform/OS: Windows 10 Problem you have experienced: I'm trying to import a custom map. After running "make import" no folder is created in CarlaUE4/content. This [...]

I am using Windows 10 with carla 0.9.13 and this issue can be easily fixed by looking at the BuildTools make script.

Looking at ./Util/BuildTools/Windows.mk (which is the file where make * command looks for corresponding OS commands), it seems that the path of the Import script is called without the proper "python" or "py" argument. It probably fallbacks to whatever default editor is set for .py files, which may be why import.py is opened in Visual Studio in @Hanss8 's case:

`import: server
	@"${CARLA_BUILD_TOOLS_FOLDER}/Import.py" $(ARGS)`

Adding `python` in front of the script path seems to do the trick for me (for some reason the path variable {CARLA_BUILD_TOOLS_FOLDER} wrote the root path name twice in my case, so I simply changed it to a relative path name instead but that probably isn't a problem for others):

`import: server
	python ./Util/BuildTools/Import.py $(ARGS)`

zhi-hui-lai, 2022-10-12:
I got it to work by changing the default python file type from Visual Studio Code to Python 
[image]