carla-simulator/carla #5358: Compile CarlaUE4.cpp 8 warnings and 14 errors generated.

ll7, 2022-04-11:
Error while working on #5347 .

[error2.txt](https://github.com/carla-simulator/carla/files/8464594/error2.txt)

ll7, 2022-04-11:
- [ ] check if Unreal 4.26 compiles correctly.

ll7, 2022-04-11:
Editor with Unreal 4.26.2 starts correctly with https://github.com/carla-simulator/carla/issues/5347#issuecomment-1092601612

ll7, 2022-04-11:
```shell
/git/carla-scoomatic/carla$ git status
On branch dev
Your branch is up to date with 'upstream/dev'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   README.md

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	Unreal/CarlaUE4/.ignore
	Unreal/CarlaUE4/Config/OptionalModules.ini

no changes added to commit (use "git add" and/or "git commit -a")
```

ll7, 2022-04-11:
BuildPythonAPI.sh: Success!

ll7, 2022-04-11:
Same error again after using `make launch`

```shell
In file included from /media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:23:
In file included from /media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:2:
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:56: error: unknown class name 'FHierarchicalStaticMeshSceneProxy'; did you mean 'FTaggedHierarchicalStaticMeshSceneProxy'?
class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
                                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                       FTaggedHierarchicalStaticMeshSceneProxy
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:7: note: 'FTaggedHierarchicalStaticMeshSceneProxy' declared here
class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
      ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:56: error: base class has incomplete type
class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
                                                ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:82:7: note: definition of 'FTaggedHierarchicalStaticMeshSceneProxy' is not complete until the closing '}'
class FTaggedHierarchicalStaticMeshSceneProxy : public FHierarchicalStaticMeshSceneProxy
      ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.h:87:35: error: 'GetViewRelevance' marked 'override' but does not override any member functions
  virtual FPrimitiveViewRelevance GetViewRelevance(const FSceneView * View) const override;
                                  ^
In file included from /media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Carla/Module.Carla.1_of_3.cpp:23:
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:170:10: error: cannot initialize return object of type 'FPrimitiveSceneProxy *' with an rvalue of type 'FTaggedHierarchicalStaticMeshSceneProxy *'
                return new FTaggedHierarchicalStaticMeshSceneProxy(MeshComponent, bIsGrass, GetWorld()->FeatureLevel, TaggedMID);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:237:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
  ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
                                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:265:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
  ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
                                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:291:97: error: no member named 'NotDrawTaggedComponents' in 'FEngineShowFlags'
  ViewRelevance.bDrawRelevance = ViewRelevance.bDrawRelevance && !View->Family->EngineShowFlags.NotDrawTaggedComponents;
                                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:300:5: error: member initializer 'FHierarchicalStaticMeshSceneProxy' does not name a non-static data member or base class
  : FHierarchicalStaticMeshSceneProxy(bInIsGrass, Component, InFeatureLevel)
    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:305:3: error: use of undeclared identifier 'bVerifyUsedMaterials'
  bVerifyUsedMaterials = false;
  ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:307:8: error: unknown type name 'FLODInfo'
  for (FLODInfo& LODInfo : LODs) {
       ^
7 warnings and 3 errors generated.
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:307:28: error: use of undeclared identifier 'LODs'
  for (FLODInfo& LODInfo : LODs) {
                           ^
/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/TaggedComponent.cpp:308:10: error: use of undeclared identifier 'FLODInfo'
    for (FLODInfo::FSectionInfo& SectionInfo : LODInfo.Sections) {
         ^
[... cut ...]
                                                                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
4 warnings generated.
7 warnings generated.
7 warnings generated.
9 warnings generated.
7 warnings generated.
10 warnings and 14 errors generated.
34 warnings generated.
7 warnings and 4 errors generated.
make[1]: *** [Makefile:315: CarlaUE4Editor] Error 6
make[1]: Leaving directory '/media/luttkule/HDD/git/carla-scoomatic/carla/Unreal/CarlaUE4'
make: *** [Util/BuildTools/Linux.mk:7: launch] Error 2
```

ll7, 2022-04-11:
no promising search result https://www.google.com/search?q=error%3A+unknown+class+name+%27FHierarchicalStaticMeshSceneProxy%27+carla&oq=error%3A+unknown+class+name+%27FHierarchicalStaticMeshSceneProxy%27+carla&aqs=chrome..69i57j69i58.1728j0j7&sourceid=chrome&ie=UTF-8

ll7, 2022-04-11:
No idea how to proceed from this point.

fu1199, 2022-06-13:
Hey! I think you did not use the Carla fork during Unreal Engine installation. I missed this step in the installation (... :) ... ) and after fixing my mistake everything compiled and launched as it should.

ll7, 2022-06-14:
Thank you, I will have a look.

ll7, 2022-06-14:
I don't know what changed, but it was now possible to launch the editor.

closed 2022-06-14 by ll7

ll7, 2022-06-14:
I might have used the wrong `UE4_ROOT`, but I doubt that. I set this variable manually.

Alphabaijinde, 2022-10-17:
> I might have used the wrong `UE4_ROOT`, but I doubt that. I set this variable manually.

git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26

Carla has its UnrealEngine fork source branch, please check it now