carla-simulator/carla #5359: SUMO Co-Simulation vehicle dynamics are not correct

p-testolina, 2022-04-11:
Hi,

CARLA version: any
Platform/OS: (SUMO/OS)-versions that were tested: 1.12/ubuntu 18.04, 1.4.0/ubuntu 20.04

We are trying to use SUMO and Traci for traffic management and CARLA for the sensor aquisition, but the angular position of the vehicle is somewhat strange. As you can see from this picture
[image]

the yaw of the vehicle does not have a smooth evolution (see zoomed picture:)
[image]

The pitch, which should be constant and always equal to 0, also presents some discontinuities that are definitely noticeable in CARLA:
[image]

I have tried different car-following models (IDM, EIDM, smartSK), step lengths, integration methods (default Euler and ballistic), and action step lengths. The issue is always present.

I attach the files to reproduce the problem (and the graphs, remove the corresponding lines if not needed)
- _E-N_Crossing_CropField.rou.xml_ contains the "ego" vehicle route, which is the only vehicle present in the simulation
- _sumo_design.sumocfg_ is the SUMO configuration file
- _Town07_Opt.net.xml_ is the SUMO network for the map. It is automatically generated from a XODR file available from the CARLA Simulator (included in the zip file as _Town07_Opt.xodr_) using the script _netconvert_carla.py_, also included.
- _test_sumo_ego_only.py_ is the Python script using Traci for the simulation.

At this point, I can't tell if I'm looking at the wrong settings, or if there is an issue with the net file. I'm fairly new to SUMO so it may be either:)

Thank you for your help!
[mwe.zip](https://github.com/carla-simulator/carla/files/8466210/mwe.zip)

fangchaooo, 2022-06-22:
The carla transfor sumo pos using below code
```
def get_carla_transform(in_sumo_transform, extent):
        """
        Returns carla transform based on sumo transform.
        """
        offset = BridgeHelper.offset
        in_location = in_sumo_transform.location
        in_rotation = in_sumo_transform.rotation

        # From front-center-bumper to center (sumo reference system).
        # (http://sumo.sourceforge.net/userdoc/Purgatory/Vehicle_Values.html#angle)
        yaw = -1 * in_rotation.yaw + 90
        pitch = in_rotation.pitch
        out_location = (in_location.x - math.cos(math.radians(yaw)) * extent.x,
                        in_location.y - math.sin(math.radians(yaw)) * extent.x,
                        in_location.z - math.sin(math.radians(pitch)) * extent.x)
        out_rotation = (in_rotation.pitch, in_rotation.yaw, in_rotation.roll)

        # Applying offset sumo-carla net.
        out_location = (out_location[0] - offset[0], out_location[1] - offset[1], out_location[2])

        # Transform to carla reference system (left-handed system).
        out_transform = carla.Transform(
            carla.Location(out_location[0], -out_location[1], out_location[2]),
            carla.Rotation(out_rotation[0], out_rotation[1] - 90, out_rotation[2]))

        return out_transform
```

namdre, 2022-10-10:
The yaw is mostly a function of the network geometries which can be made smoother by setting netconvert option **--junctions.corner-detail** (i.e. *10* instead of the default *5*).  Depending on the simulation option, lane changing operations can influence the yaw at might be another cause of discontinuities (https://sumo.dlr.de/docs/Simulation/SublaneModel.html)

The pitch comes directly out of the network geometry at the vehicle front and is currently not interpolated between front and back of the vehicle (a known issue in SUMO)