carla-simulator/carla #5365: Navigating ego vehicle through custom waypoints

TR-Monash, 2022-04-13:
I'm trying to simulate some near-collision instances such as two vehicles passing close by. To do so, my plan was to setup two PID-controlled ego vehicles rigged with cameras navigating through custom waypoints in two adjacent lanes. These waypoints were to based on the waypoints already on the map and modified so that one set of waypoints comes closer to the other, as shown below.

```       while True:
            waypoints = world.get_map().get_waypoint(vehicle.get_location())
            waypoint = np.random.choice(waypoints.next(0.5))
            waypoint.transform.location.y += 1.5
```

However, it seems that the waypoints are static and cannot be modified (they are always right in the middle of a lane) as their transform is not affected as expected. Is there any other way of achieving this?
Thanks.

closed 2022-04-20 by TR-Monash

JunadMQ, 2022-06-15:
> I'm trying to simulate some near-collision instances such as two vehicles passing close by. To do so, my plan was to setup two PID-controlled ego vehicles rigged with cameras navigating through custom waypoints in two adjacent lanes. These waypoints were to based on the waypoints already on the map and modified so that one set of waypoints comes closer to the other, as shown below.
> 
> ```
>             waypoints = world.get_map().get_waypoint(vehicle.get_location())
>             waypoint = np.random.choice(waypoints.next(0.5))
>             waypoint.transform.location.y += 1.5
> ```
> 
> However, it seems that the waypoints are static and cannot be modified (they are always right in the middle of a lane) as their transform is not affected as expected. Is there any other way of achieving this? Thanks.

"  Having the same issue as you described. May I know how did you resolve it. This is the link to my posted issue:https://github.com/carla-simulator/carla/issues/5516

TR-Monash, 2022-06-26:
@JunadMQ Sorry, wasn't able to do this. Tuning the PID parameters was the best I could do.