carla-simulator/carla #5367: instance segmentaion bug in town01

zxiaomzxm, 2022-04-15:
CARLA version: 0.19.3
Platform/OS: ubuntu 20.04
Problem you have experienced: 
I use instance segmentation camera in **town01**, but capture strange results:
rgb image
[image]
instance image
[image]
It seems that the instance camera can not see the mound in the right side and ray casting just pass through it.
Other maps has no such phenomenon.

XGodina, 2022-05-24:
Hi @zxiaomzxm, You wrote "CARLA version: 0.19.3", do you mean the " CARLA version 0.9.13", right?. I will take a look. Thank you for letting us know

zxiaomzxm, 2022-06-01:
> Hi @zxiaomzxm, You wrote "CARLA version: 0.19.3", do you mean the " CARLA version 0.9.13", right?. I will take a look. Thank you for letting us know

yep, it's 0.9.13. And I found town04 have similar problem.
[image]
[image]

ghh116553, 2022-06-22:
> Hi @zxiaomzxm, You wrote "CARLA version: 0.19.3", do you mean the " CARLA version 0.9.13", right?. I will take a look. Thank you for letting us know

I meet the same issue. InstanceCamera can not show the landscape correctly.

JosenJiang, 2023-04-20:
@zxiaomzxm Hi, you seem to have implemented an instance segmentation fisheye camera in Carla. Would you mind sharing your methods? I had some problems implementing this feature.

AreopagX, 2024-09-11:
@Blyron Can you give hints how this might be resolvable?

tdegraaff, 2025-01-30:
@jorge-kabuto Can now be closed (https://github.com/carla-simulator/carla/pull/8528).