carla-simulator/carla #5371: make package error on windows

zhangn3x, 2022-04-19:
I've always had no problem packing my carla release package, but when I want to test my new tag I find it is error for UE4Editor in make package(carla:0.9.13 UE4:4.26.):
[Upgrade]
Building 30 actions with 8 processes...
  [1/30] SharedPCH.UnrealEd.cpp
  [2/30] PCH.CarlaUE4.cpp
  [3/30] Module.Carla.gen.4_of_7.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  [4/30] Module.Carla.gen.3_of_7.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  [5/30] Module.Carla.gen.2_of_7.cpp
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(61): warning C4996: 'AWheeledVehicle': PhysX is deprecated. Use the AWheeledVehiclePawn from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/CarlaWheeledVehicle.h(161): warning C4996: 'UWheeledVehicleMovementComponent4W': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent from the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(53): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Game/CarlaHUD.h(54): warning C4996: 'UWheeledVehicleMovementComponent': PhysX is deprecated. Use the UChaosWheeledVehicleMovementComponent fron the ChaosVehiclePhysics Plugin. Please update your code to the new API before upgrading to the next release, otherwise your project will no longer compile.
  [6/30] Module.Carla.2_of_3.cpp
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(84): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(84): error C2143: syntax error: missing ';' before '&'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(84): error C2065: 'Registry': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(84): error C2027: use of undefined type 'UCarlaEpisode'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaRecorder.h(37): note: see declaration of 'UCarlaEpisode'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(87): error C2065: 'Registry': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(87): error C3536: 'It': cannot be used before it is initialized
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(87): error C2065: 'Registry': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(91): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(94): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(94): error C2065: 'Other': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(99): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(99): error C2065: 'Vehicle': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(110): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(110): error C2065: 'Walker': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(120): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(120): error C2065: 'TrafficLight': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(129): error C2027: use of undefined type 'UCarlaEpisode'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaRecorder.h(37): note: see declaration of 'UCarlaEpisode'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(132): error C2027: use of undefined type 'UCarlaEpisode'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaRecorder.h(37): note: see declaration of 'UCarlaEpisode'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(152): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(156): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(155): error C2440: '<function-style-cast>': cannot convert from 'initializer list' to 'CarlaRecorderPosition'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(155): note: Invalid aggregate initialization
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(166): error C2027: use of undefined type 'FCarlaActor'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Recorder\CarlaReplayerHelper.h(20): note: see declaration of 'FCarlaActor'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(171): error C2065: 'FVehicleControl': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(171): error C2146: syntax error: missing ';' before identifier 'Control'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(171): error C2065: 'Control': undeclared identifier
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Recorder/CarlaRecorder.cpp(172): error C2027: use of undefined type 'FCarlaActor'
[... cut ...]
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(193): error C2027: use of undefined type 'ACarlaWheeledVehicle'
  D:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla/Vehicle/WheeledVehicleAIController.h(18): note: see declaration of 'ACarlaWheeledVehicle'
  D:/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/YieldSignComponent.cpp(204): error C2027: use of undefined type 'ACarlaWheeledVehicle'

Previously, this problem still existed in my make launch process, so I try to rebuild my C++, UE4 and carla, but now it will still appear in the make package process

wkgud9090, 2022-07-19:
anyone solve this issue?

TR-Monash, 2022-07-19:
I get similar errors for `CarlaRecorder.cpp` on one of my computers, but it works perfectly fine on the other computer I have!
Tried many different things (ranging from confirming environmental variable to changing the code) to no avail.

Rocky0102, 2022-12-05:
I met this annoying problem several times, exactly the same error messages, and tried many ways and  finally was able to solve it. 
I deleted everything under folder carla\Unreal\CarlaUE4\  except Content and Config, then repull those deleted files/folders from github, and recompiled.
I think maybe the reason was that I generated VS solution project files with CarlaUE4.uproject and opened it with VS2019, the two building systems may have something conflict.

Blyron, 2024-05-03:
We cannot reproduce this issue, could you make clean and then try again? 
If you have solved the issue are you happy to close it?

Blyron, 2024-05-17:
We are closing this issue as we have no responde feel free to reopen if something comes up

closed 2024-05-17 by Blyron