carla-simulator/carla #5377: when would CARLA support Unreal Engine 4.27

shengkaicui, 2022-04-21:
Dear CARLA development team, 

Since Carsim 2022.0 UE plugin only support Unreal Engine 4.27, while CARLA 0.9.13 use Unreal Engine 4.26, which lead to UE version mismatching and Carsim could not integrated with CARLA.
When would upgrade Unreal Engine in CARLA to 4.27?

Thanks!

shengkaicui, 2022-04-27:
i rebuild CarSim UE plugin in UE editor, so the CarSim 2022.0 could be integrated in CARLA 0.9.13 now, i would close this issue.

closed 2022-04-27 by shengkaicui

sushil-bhatta, 2022-12-19:
Hi @shengkaicui,
could you please tell in details, how do you solved this issue. I am a beginner and haven't worked till yet with Unreal Engine. So it would be nice, if you could give some tips.

pattersonchien, 2023-01-18:
How to rebuild the CarSim plugin to make it compatible with Carla Engine? I use the engine version that carla build in windows 10. the carla version is 0.9.14.
[image]