carla-simulator/carla #5382: How to change default map when make launch?

Utopialjp, 2022-04-22:
How to change default map when run "make launch"?

closed 2022-04-28 by Utopialjp

kubs0ne, 2023-02-07:
Did you manage to find the answer?

Daraan, 2023-10-11:
This is one of the first search results coming up for this question.

Has someone found a solution?

command-z-z, 2023-10-28:
I don't know how to change default map by `make launch` command. But another way can also do it, when you launch carla in UE. You can click the menu in bottom, Carla->Maps->Town01 to load the Town01 map.

Daraan, 2023-10-30:
I found the answer in `<Carla>/Unreal/CarlaUE4/Config/Default.Engine.ini`

there are the following settings:

```
[/Script/EngineSettings.GameMapsSettings]
EditorStartupMap=/Game/Carla/Maps/Town04.Town04
GameDefaultMap=/Game/Carla/Maps/Town04.Town04
ServerDefaultMap=/Game/Carla/Maps/Town04.Town04
GlobalDefaultGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
GameInstanceClass=/Script/Carla.CarlaGameInstance
TransitionMap=/Game/Carla/Maps/Town04.Town04
GlobalDefaultServerGameMode=/Game/Carla/Blueprints/Game/CarlaGameMode.CarlaGameMode_C
```

I did not look deeper into the 4 different settings, especially not TransitionMap.  
I changed the Town10_Opt vs Town04 and everything `make launch` loaded Town04 without problems.

tarynnielsen, 2024-06-27:
After you run `make launch` go to Edit->Project Settings

Once you're there under 'Project' go to 'Maps & Modes' and there you can change the default Editor Startup Map and the Game Default Map.

Then if you exit and rerun `make launch` it should load the new default map that you chose.