carla-simulator/carla #5386: could not find any instance of Visual Studio when making PythonAPI

tonyliu02, 2022-04-24:
Platform: 
- Windows 10
- cmake version: 3.23.1
- Visual Studio version: v15.9.46

When I ran the command "make PythonAPI", the following error occured:
[image]
it seems like it is an issue with the version of my Visual Studio, but I am not sure. The VS I installed was VS2017 v15.9.46. Did anyone encounter the same problem or know how to fix it? Thanks in advance!

OmarElabd, 2022-04-26:
Looks like you need to install VS 2019

nicolasrosa, 2025-02-24:
@tonyliu02 I solved the mentioned problem by editing line 68 from the `Util\BuildTools\Setup.bat` file.

I changed the line from:
`if %GENERATOR% == "" set GENERATOR="Visual Studio 16 2019"`

to:

`if %GENERATOR% == "" set GENERATOR="Visual Studio 17 2022"`

Blyron, 2025-02-24:
How to build CARLA UE4 with vs 2022 https://docs.google.com/document/d/1JmHQiLii-wehCe0JyHSpvRiI54i9IaxPLOgWSeHb6Mk/edit?tab=t.0

closed 2025-02-24 by Blyron