carla-simulator/carla #5391: More parked cars?

truncs, 2022-04-27:
CARLA version: master
Platform/OS: Linux
Problem you have experienced: More parked cars in a map
What you expected to happen: Configurable variable that let's me add more number of parked cars
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

Is there a way to add more parked cars to a map in Carla? In the real world the distribution of parked cars is different from what is seen in the maps.

MrLerry, 2022-05-02:
I guess you could just write a small script which spawns a car without an autopilot and give them coordinates where you want to have them spawned.
something like this:
`try:
        ego_vehicle = world.get_blueprint_library().filter('vehicle.audi.etron')[0] 
        EGOtransf = carla.Transform(carla.Location(x=TPosX, y=TPosY, z=TPosZ), carla.Rotation())
        SpawnEgo = world.spawn_actor(ego_vehicle, EGOtransf)
        print("\nSpawning Target at ", EGO_sp)
        input("\nPress Enter to terminate the Target")
finally:
        lidar_sen.destroy()
        client.apply_batch([carla.command.DestroyActor(x) for x in Props])`

XGodina, 2022-05-24:
Hi @truncs, 
Right now we don't have any script to do that. But we have it in mind for future releases.

closed 2022-05-24 by XGodina