carla-simulator/carla #5393: "make package" with custom maps included

MrLerry, 2022-04-27:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04

I integrated a large map with the UE4-Editor and put a lot of effort into editing it. 
For better performance and for sharing a ready build Carla-version with my colleges I wanted to package my progress for them to use the new map as well, but the "make-package" command didn't integrate the new map into the now "shipping" version under /Dist/

What did i do wrong? Do i need to edit the make-Files?
The assets i integrated as well are there but can't be found with an "index out of range error". That leads me to the supposition that I located my stuff in wrong directories? 

Any help would be appreciated.

Cheers,
Lerry

MrLerry, 2022-05-02:
Found it, just add the Path to the maps you want to add into your package under "Edit/ProjectSetting/Packaging". The default maps are there.
then execute "make package" as the docs suggest

closed 2022-05-02 by MrLerry

IrfanAli91170, 2025-02-21:
If it's a map divided in tiles, how to include these or we have to include only the base level ?