carla-simulator/carla #5395: Want to know how to measure distance of detected objects using camera and radar sensor in CARLA

sbv198, 2022-04-28:


sbv198, 2022-04-28:
I have used camera RGB and radar i am getting vector 3D of each pixels generated by RADAR and for that i have used the code provided in carla documentation. i want to measure distance from an object using the image captured by camera. how to do that

any help is appreciated

thanks
Sagar Varvariya

PatrickFSLin, 2022-04-29:
the same question here...

AmirMz1, 2022-08-21:
Hello, thank you for your answer
I want to get the distance to each car.
Can you please guide me how to do this, what the radar sensor detects and how to set it only for cars.

> @PatrickFSLin I used following code to measure average distance of detected objects. the following code you will find it on carla documentation as well.
> 
> 1. https://carla.readthedocs.io/en/latest/tuto_G_retrieve_data/#set-advanced-sensors
> 2. https://carla.readthedocs.io/en/latest/ref_sensors/#depth-camera
> 
> refer this documents as well
> 
> # RADAR sensor
> ```
>     rad_bp = world.get_blueprint_library().find('sensor.other.radar')
>     rad_bp.set_attribute('horizontal_fov', str(35))
>     rad_bp.set_attribute('vertical_fov', str(20))
>     rad_bp.set_attribute('range', str(20))
>     rad_location = carla.Location(x=2.0, z=1.0)
>     rad_rotation = carla.Rotation(pitch=5)
>     rad_transform = carla.Transform(rad_location, rad_rotation)
>     rad_ego = world.spawn_actor(rad_bp, rad_transform, attach_to=ego_vehicle,
>                                 attachment_type=carla.AttachmentType.Rigid)
>     actor_list.append(rad_ego)
> 
>     def rad_callback(radar_data):
>         velocity_range = 7.5  # m/s
>         current_rot = radar_data.transform.rotation
>         for detect in radar_data:
>             azi = math.degrees(detect.azimuth)
>             alt = math.degrees(detect.altitude)
>             # The 0.25 adjusts a bit the distance so the dots can
>             # be properly seen
>             fw_vec = carla.Vector3D(x=detect.depth - 0.25)
>             carla.Transform(
>                 carla.Location(),
>                 carla.Rotation(
>                     pitch=current_rot.pitch + alt,
>                     yaw=current_rot.yaw + azi,
>                     roll=current_rot.roll)).transform(fw_vec)
> 
>             def clamp(min_v, max_v, value):
>                 return max(min_v, min(value, max_v))
> 
>             norm_velocity = detect.velocity / velocity_range  # range [-1, 1]
>             r = int(clamp(0.0, 1.0, 1.0 - norm_velocity) * 255.0)
>             g = int(clamp(0.0, 1.0, 1.0 - abs(norm_velocity)) * 255.0)
>             b = int(abs(clamp(- 1.0, 0.0, - 1.0 - norm_velocity)) * 255.0)
>             world.debug.draw_point(
>                 radar_data.transform.location + fw_vec,
>                 size=0.075,
>                 life_time=0.1,
>                 persistent_lines=False,
>                 color=carla.Color(r, g, b))
> ```
> 
> # Array tranfer of obtained data
> ```
>             points = np.frombuffer(radar_data.raw_data, dtype=np.dtype('f4'))
>             points = np.reshape(points, (len(radar_data), 4))
> 
>             print(points)
> 
>  # code convert array into list and measure distance
>             L = []
>             pointslist=points.tolist()
>             for i in range(len(pointslist)):
>                 L.append(pointslist[i-1][-1])
> 
>             ave = sum(L)/len(L)
>             print(ave)
> 
>     rad_ego.listen(lambda radar_data: rad_callback(radar_data))
>     print(rad_ego.listen)
> ```
> 
> good luck

SExpert12, 2023-07-18:
> Hello, thank you for your answer I want to get the distance to each car. Can you please guide me how to do this, what the radar sensor detects and how to set it only for cars.

have you done this part where you can get the distance of each car?