carla-simulator/carla #5399: Vehicle Not Moving In CARLA

anandkvmar, 2022-04-29:
Hey everyone,

I’m working on importing my own vehicle model into Carla, but I’m facing an issue while running the vehicle in the Carla simulator. The vehicle doesn’t seem to be moving and the wheels are movies in a different axis. I have attached the relevant images and video of the issue. Can somebody please look into the issue and help me find a fix.
YouTube link for my issue: https://youtu.be/3E--y-spoiQ

CARLA version: CARLA 0.9.13
Platform/OS: Ubuntu 20.04 LTS
Problem you have experienced: Vehicle not moving in the carla simulator.
What you expected to happen: The vehicle to function properly.
Steps to reproduce: Referred the CARLA documentation.
Other information (documentation you consulted, workarounds you tried): https://github.com/carla-simulator/carla/issues/2738#issuecomment-618681769

nikitaverbis, 2022-05-01:
Hi, [anandkvmar](https://github.com/anandkvmar)!
The problem possibly in collision box and spheres of your model. You probably have an overlap of collisions or a problem in the wheel modeling.
Please run simulation for your vehicle blueprint and check collision settings as suggested in tutorial for importing vehicles, for example, https://www.youtube.com/watch?v=iTjxkgjRbfY

anandkvmar, 2022-05-02:
@nikitaverbis, Thanks for replying. I'll try solving the issue by following the video and later give an update on it.

XGodina, 2022-05-24:
Hi @anandkvmar, So far I see it seems everything it's ok. But I have some questions that maybe help.
Do you configure the blueprint of the car ? There is a part on set the wheels maybe you forgot that.
https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/ (Point 9)
Also, I see in your video. That is the wheel blueprint you have to change the Collision Mesh to Wheel_Shape.
Let me know if it works

anandkvmar, 2022-05-24:
Ok, I will check it out.

kennyS99, 2022-07-18:
Hey anandkvmar,
I think the problem of the wheels move on a different axis is you did not set constrain in the blender software.
and my model cannot move just like you and the wheels are rotating. Have  you solved your problem?

anandkvmar, 2022-07-18:
Hey @kennyS99 ,
So I have tried out the procedures as  https://github.com/carla-simulator/carla/issues/2738#issuecomment-618681769
Other than that I also referred to this video https://youtu.be/_ZBQg7293v0
In the YouTube video, the constrains were different than then once shown in the carla guide. My vehicle moves now but the vehicle physics aren't that good. 
As of now, I'm not working in this project but we did get the vehicle to move in the map.

kennyS99, 2022-07-19:
Hey anandkvmar,
I am appreciate your quick response.
I followed your shared video in the youtube. But my car still cannot move, only the wheels rotate. Have you guys figured the problem out? I even do not know where is the problem?
Thx

anandkvmar, 2022-07-21:
Hey @kennyS99 ,
Try to see it the all the 4 wheels and vehicle body are linked to the skeleton properly for this, make sure the car is rigged. For rigging I used a blender addon, watch this video https://youtu.be/laAs3lefrKY

 We couldn't work further on our project because unreal engine used to crash after like 10 sec of running the project file (simulation). We also didn't have a high spec system.

closed 2022-07-24 by anandkvmar