carla-simulator/carla #5405: Longitudinal and lateral velocity calculation

sbv198, 2022-05-03:


KevinLADLee, 2022-05-09:
This code piece is only for my usage, so that it may not be the simplest way. But the principle is the same.
That is "apply the inverse rotation matrix of the vehicle to the velocity vector". 
Left top 3x3 of the transform matrix (4x4) is the rotation matrix. (That is SE(3).)

You could directly use this equation.
[image]

```python
        trans_mat = numpy.array(carla_vehicle.get_transform().get_matrix()).reshape(4, 4)
        rot_mat = trans_mat[0:3, 0:3]
        inv_rot_mat = rot_mat.T
        vel_vec = numpy.array([carla_velocity_vec3.x,
                                              carla_velocity_vec3.y,
                                              carla_velocity_vec3.z]).reshape(3, 1)
        vel_in_v = inv_rot @ vel_vec
```
vel_in_v[0] is the vx in the vehicle frame, that is, the longitudinal speed of the vehicle.

sbv198, 2022-05-10:
@KevinLADLee  Thanks a lot for your explanation. 

I used method shown in code below to get velocities, i was succesfully able to retrieve data.   

ego_vehicle_vel = ego_vehicle.get_velocity()
ego_vehicle_velocity = np.array([ego_vehicle_vel.x, ego_vehicle_vel.y, ego_vehicle_vel.z])
ego_vehicle_vel_longitudinal = abs(ego_vehicle_velocity[0])

But i had some issues like i was getting some values in negative as high as -5.66 m/s so that's why i used abs command to get absolute value.

You can correct if you think i made any mistakes?