carla-simulator/carla #5429: Building carla on windows running "make PythonAPI" encounter issue with zlib

RichardBaoLei, 2022-05-10:
I ran into this error while building carla on windows, trying to search for solution but got no luck.
here is the error I got

[image]

the problem seems to be zlib library cannot be found, but i do have zlib(manually download 1.2.11 or change version to 1.2.12 in cmakelists)

i am not sure what i should do now to solve this problem, been stuck on this for few days is really painful.
any suggestions would be appreciated!

bjoernbahn, 2022-05-11:
Seems like zlib has updated its library and is not longer offering 1.2.11 for download.

Fix, which worked for me:
Change in Util/InstallersWin/install_zlib.bat row 51 from 1.2.11 to 1.2.12
Delete the Build Directory manually and try again.

RichardBaoLei, 2022-05-12:
> 

As I mentioned I have already tried that and it's not working for me though.
none of 1.2.11 and 1.2.12 version work. 
Any other suggestions?

XGodina, 2022-05-24:
Hi@RichardBaoLei @bjoernbahn, I think you need to update the xerces library and compile it again.

bernatx, 2022-05-24:
RichardBaoLei, check your **antivirus**. Probably it is preventing to download the zlib file.
Check for an error like: **This script contains malicious content and has been blocked by your antivirus software.**