carla-simulator/carla #5430: CARLA and Simulink Interface

sbv198, 2022-05-10:
CARLA version: 0.9.13
Platform/OS: Windows 10

Is there anyone who has done interface between carla and simulink , I need help with something.  My problem is i have managed to extract data from carla and managed to send it in matlab workspace  but there is  a thing that it only takes one value at a time, like we receive some information from CARLA at each istance so what it does is it erases last information and assigned new information in workspace. Any idea?

Thanks in advance

Reyen994, 2022-05-16:
can i ask how you managed to extract the data? I would like to deal with that right now

sbv198, 2022-05-16:
@Reyen994 I hope that given answer would be helpful 

 i think we should help each other since both of us are working on the similar things!!

Reyen994, 2022-05-16:
Yes it's great that you're working on it too. 
So how do I start Matlab/simulinke and load I understood. Only not yet how I send values to Matlab (or Simulink). I only have a simple input with a scope in Simulink for now. There I would like to represent the vehicle speed after the time. Do you have an idea how this works?

I am currently working on integrating CarSim better into Carla and am now trying to do this via Simulink (since the provided plugin is not very good). 

Translated with www.DeepL.com/Translator (free version)

Reyen994, 2022-05-16:
Yes it's great that you're working on it too. 
So how do I start Matlab/simulinke and load I understood. Only not yet how I send values to Matlab (or Simulink). I only have a simple input with a scope in Simulink for now. There I would like to represent the vehicle speed after the time. Do you have an idea how this works?

I am currently working on integrating CarSim better into Carla and am now trying to do this via Simulink (since the provided plugin is not very good).

sbv198, 2022-05-16:
@Reyen994 
I also want to know how to create velocity map with respect to time but i couldn't do it for the moment.
 one reason being is that carla do simulation every time like it does one simulation then it wait for another simulation so that's why it will send only one value at a time not like a  in the form of array or anything for instance i know only till that.

if i have more information i will share

sbv198, 2022-05-17:
@Reyen994  tell me something, I am using following command to load a world but i am having some error, can you tell me whats wrong over here in the code,

world.load_world('Town05')
AttributeError: 'World' object has no attribute 'load_world'

sbv198, 2022-05-24:
@Reyen994 i wanted to ask have you managed to plot graphs of velocity and acceleration with respect to time? 
if yes, could you share it ?

thanks

closed 2022-06-15 by sbv198

markpriyanshu, 2024-03-14:
hello, I am also trying to integrate simulink model with carla. I tred runnung @sbv198 code but the vehicle is not running correctly. Can u pls show your simulink model