carla-simulator/carla #5431: Only one floor mesh when importing the map from roadrunner

javiercalle97, 2022-05-11:
Hello! I'm trying to import a Roadrunner map manually following this tutorial https://carla.readthedocs.io/en/latest/tuto_M_add_map_alternative/#manual-import (manually because I'm having some problems with make import). And i'm facing a problem. When the map is imported into Unreal the floor is only one mesh so i can't identify sidewalks, roads, etc because it is only one mesh :( 
Someone has suffered this? 
(I have tried with different maps and always happen the same).

Thanks you in advance!

Carla version: 9.13

NOTE: I always checked that the "combine meshes" option is disabeled

XGodina, 2022-05-24:
Hi @javiercalle97 
I think the problem is when you export the .fbx. In RoadRunner. You must choose split geometry by Semantic Segmentation. This should work. Let me know if it works or not
[image]

javiercalle97, 2022-06-03:
Yes thanks! It worked. :)