carla-simulator/carla #5435: FBX SDK

ll7, 2022-05-13:
I have the same issue while building carla in a docker:

```shell
BuildPythonAPI.sh: Success!
BuildUtilsDocker.sh: Downloading FBX SDK...
--2022-05-13 06:26:31--  https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz
Resolving www.autodesk.com (www.autodesk.com)... 104.125.70.182
Connecting to www.autodesk.com (www.autodesk.com)|104.125.70.182|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2022-05-13 06:26:31 ERROR 403: Forbidden.

make: *** [build.utils] Error 8
Util/BuildTools/Linux.mk:149: recipe for target 'build.utils' failed
The command '/bin/sh -c cd /home/carla/ &&   if [ -z ${GIT_BRANCH+x} ]; then git clone --depth 1 https://github.com/carla-simulator/carla.git;   else git clone --depth 1 --branch $GIT_BRANCH https://github.com/carla-simulator/carla.git; fi &&   cd /home/carla/carla &&   ./Update.sh &&   make CarlaUE4Editor &&   make PythonAPI &&   make build.utils &&   make package &&   rm -r /home/carla/carla/Dist' returned a non-zero code: 2
```

Where should I use this flag? In my docker command or do I have to modify the `BuildUtilsDocker.sh`. I use the current dev branch of carla.

```shell
docker build -t carla -f Carla.Dockerfile .
```

_Originally posted by @ll7 in https://github.com/carla-simulator/carla/issues/5342#issuecomment-1125702466_

ndeshp2s, 2022-06-21:
Hi @ll7, Could you manage to apply the fix and resolve the issue?

ll7, 2022-06-28:
No, I was not able to fix it.

l1th1um, 2022-07-25:
Its should be on Util\BuildTools\BuildUtilsDocker.sh at line 20

`wget --user-agent="Mozilla/4.0 (compatible; MSIE 6.0; Windows NT 5.1; SV1)" -c "${FBXSDK_URL}" -P "${CARLA_DOCKER_UTILS_FOLDER}"`

I haven't try if this working or not. Since it took me 6 hours to build the image until this downloading process

l1th1um, 2022-07-26:
Can confirm its work. But don't forget to change the dockerfile, so its using your local directory instead of clone from the repo

muety, 2024-02-01:
I'm having a similar issue, but instead of getting a 403, the request seemingly just never completes. It's stuck at "awaiting response..." forever. Any ideas? 

[image]

muety, 2024-02-01:
Managed to work around this by changing the above `wget` command to:
```bash
curl -O --output-dir "${CARLA_DOCKER_UTILS_FOLDER}" -L -A "Mozilla/4.0 (compatible; MSIE 6.0; Windows NT 5.1; SV1)" "${FBXSDK_URL}"
```

apolak, 2024-03-21:
Looks like the problem is caused by the use of `Mozilla` as user agent here:
https://github.com/carla-simulator/carla/blob/cead6d1b05e89ad84df80984f62daba57bb50e55/Util/BuildTools/BuildUtilsDocker.sh#L20

Change it to a realistic, up-to-date user agent like `Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36` and it will work again.

But the response is a `301 Moved Permanently` with `Location: https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz`, so you might want to update the URL instead. This way, you won't need to set the user agent at all:

```console
$ wget https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz --user-agent="Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/123.0.0.0 Safari/537.36" --max-redirect=0
--2024-03-21 09:32:59--  https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz
Resolving www.autodesk.com (www.autodesk.com)... 104.81.62.87, 2a02:26f0:d200:483::4516, 2a02:26f0:d200:490::4516
Connecting to www.autodesk.com (www.autodesk.com)|104.81.62.87|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz [following]
0 redirections exceeded.
$ wget https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz --max-redirect=0
--2024-03-21 09:33:11--  https://damassets.autodesk.net/content/dam/autodesk/www/adn/fbx/2020-0-1/fbx202001_fbxsdk_linux.tar.gz
Resolving damassets.autodesk.net (damassets.autodesk.net)... 104.81.62.87, 2a02:26f0:d200:490::4516, 2a02:26f0:d200:483::4516
Connecting to damassets.autodesk.net (damassets.autodesk.net)|104.81.62.87|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 55375460 (53M) [application/x-gzip]
Saving to: ‘fbx202001_fbxsdk_linux.tar.gz’

fbx202001_fbxsdk_linux.tar.gz             100%[==================================================================================>]  52.81M  53.0MB/s    in 1.0s    

2024-03-21 09:33:12 (53.0 MB/s) - ‘fbx202001_fbxsdk_linux.tar.gz’ saved [55375460/55375460]
```

closed 2025-08-05 by Blyron