carla-simulator/carla #5439: Build Carla and UE4 in Docker

MrLerry, 2022-05-13:
Hi,

for using some generated Assets and Maps in a packaged Carla version I wanted to Build Carla and the UE4 in docker on an external drive. 
I mounted the external drive to /var/lib/docker and Docker seems to work just fine with this. 

Now for the problem: 
When i execute `docker build --build-arg EPIC_USER=<GitHubUserName> --build-arg EPIC_PASS=<GitHubPassword> -t carla-prerequisites -f Prerequisites.Dockerfile .` an Ubuntu18 Image is build and tagged succesfully when searching for it with "docker images".
But the ~100GB UE4 Image is tagged with <none> and the commandpromt  doesn't exit after it says UE4 is build and tells me the executiontime. Thats why the the second command for building Carla fails i guess.

Has anybody encountered this Issue before?

CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04

Cheers Lerry

apgeorg, 2023-01-23:
Hi, actually no but do you have any updates on this?

MrLerry, 2023-01-24:
No i gave up on this :)

closed 2023-01-24 by MrLerry