carla-simulator/carla #5441: A mistake in the document?

Lumozz, 2022-05-15:
In document First steps/Adding NPCs:
# for each spawn point, we choose a random vehicle from the blueprint library
```
for i in range(0,50):
    world.try_spawn_actor(random.choice(vehicle_blueprints, random.choice(spawn_points)))
```
should be this?
```
for i in range(0,50):
    sp = random.choice(spawn_points)
    world.try_spawn_actor(random.choice(vehicle_blueprints), random.choice(spawn_points))
```

junjiel, 2022-09-05:
I was faced with the same issue when getting started with Carla. 
It gives me the following error:

> ---------------------------------------------------------------------------
> TypeError                                 Traceback (most recent call last)
> ~\AppData\Local\Temp\ipykernel_5748\3851441670.py in <module>
>       5 # for each spawn point, we choose a random vehicle from the blueprint library
>       6 for i in range(0,50):
> ----> 7     world.try_spawn_actor(random.choice(vehicle_blueprints, random.choice(spawn_points)))
> 
> TypeError: choice() takes 2 positional arguments but 3 were given

@Lumozz  your solution should be right.

The code in the documentation should be changed as below:
```
for i in range(0,50):
    world.try_spawn_actor(random.choice(vehicle_blueprints), random.choice(spawn_points))
```

MattRoweEAIF, 2024-04-12:
This issue has been fixed, thank you for alerting us.

closed 2024-04-12 by MattRoweEAIF