carla-simulator/carla #5445: Carla CarSim - Simulation stop

Reyen994, 2022-05-16:
CARLA version: 0.9.13
Platform/OS: Win 10
Problem you have experienced: simulation stops when the CarSim plugin is started

I installed the PlugIn from CarSim and added the interface (vehicle.enable_carsim(<path_to_ue4simfile.sim>)) in my script. The problem is that as soon as I activate CarSim with the appropriate command, the vehicle stops and does not continue (in any case not in the simulation). Does anyone know the problem or does anyone have a solution?

Reyen994, 2022-05-16:
can i ask how you managed to extract the data? I would like to deal with that right now

sushil-bhatta, 2022-12-19:
Hi @Reyen994, did you solved this issue ? I am also having this problem. If you have any progress on it, could you please share some information!