carla-simulator/carla #5447: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.

loveskycloud, 2022-05-17:
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

scenario_runner.py:94: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.12'):
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API
WARNING: Client API version     = 0.9.13
WARNING: Simulator API version  = 0.9.13-dirty
INFO:  Found the required file in cache!  Carla/Maps/TM/Town01.bin

kennyS99, 2022-08-03:
I have the same problem, my carla version is 0.9.12 and i download the package version of scenarioRunner with 0.9.12 but still have the version mismatch warning.

zhaoyongqi2022, 2022-10-27:
Hi, there,

I encounter the same problem:

Carla version: 0.9.13
Scenario runner version: 0.9.13
Error message: 
scenario_runner.py:94: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.12'):

Did you solve the problem?

ravivaghasiya1998, 2023-08-31:
scenario_runner.py:94: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if LooseVersion(dist.version) < LooseVersion('0.9.12'):
Configuration for scenario D:\CARLA_9_13\scenario_runner\srunner\scenarios\change_lane.py cannot be found!
No more scenarios .... Exiting

I am also getting the same error. did amyone has solution ?