carla-simulator/carla #5453: lidar coordinate to world coordinate transformation error

AOOOOOA, 2022-05-19:
Carla-version: 0.9.13
OS: ubuntu 18.04 

Hi, 

I have two vehicles in the Carla world, and both of them have attached a lidar. 
This is what I want to achieve:

**_object coordinate in vehicle_1 lidar coordinate system ---> world coordinate ----> corresponding coordinate in vehicle_2 lidar coordinate_**

This is my procedure:
1. I got lidar1_to_world transformation matrix using the function **_lidar.get_transform().get_matrix()_**.  Also, I got a lidar2_to_world matrix using the same function. Since I need world_to_lidar2 transformation matrix, I use the formula below to get it.
[image]

2. I got the 3D object detection result from vehicle 1 and its corresponding bbox lidar1 coordinate.(bbox_v1) The object detection model is PointRCNN in OpenPCDet.
3. I transform the bbox_v1 coordinate into bbox_world coordinate using lidar1_to_world. And then transform the bbox_world to bbox_v2 coordinate using the world_to_lidar2 transformation matrix.

However, after finishing the whole process, I plot the transformed bbox in point cloud of vehicle 2. I found out that the bbox_v2 has a considerable deviation shown below. The vehicle in red circle is vehicle_1.
 
[image]

 Does there any error in the process or do I have any misunderstanding about the coordinate transformation in Carla?

AOOOOOA, 2022-05-19:
I found out that is because of  typo in my code and it is ok now. 
[image]

closed 2022-05-19 by AOOOOOA

makaveli10, 2022-05-31:
@AOOOOOA Hi, I am trying to work on creating a synthetic dataset out of CARLA in the kitti dataset. And I am stuck in this issue https://github.com/open-mmlab/mmdetection3d/issues/1512. However, it seems that you have been able to create a 3d object detection dataset out of CARLA. If you could give me some pointers that would be great. 
Thanks

AOOOOOA, 2022-06-01:
> @AOOOOOA Hi, I am trying to work on creating a synthetic dataset out of CARLA in the kitti dataset. And I am stuck in this issue [open-mmlab/mmdetection3d#1512](https://github.com/open-mmlab/mmdetection3d/issues/1512). However, it seems that you have been able to create a 3d object detection dataset out of CARLA. If you could give me some pointers that would be great. Thanks

I didn't totally get your stuck point. Where did you get your dataset?  It seems like you stuck at the lidar coordinate and camera coordinate transformation. There are some tools for visualizing the KITTI dataset, they have the function to project the lidar point on the image. You can use this function to check whether your transformation matrix is right or not.

makaveli10, 2022-06-01:
@AOOOOOA Yes, lidar and camera coordinate frame. As for the code I am writing it myself so, you mean tools like meshlab? or do you have reference to any other tool that might help project lidar points on image. 
thanks

AOOOOOA, 2022-06-01:
> @AOOOOOA Yes, lidar and camera coordinate frame. As for the code I am writing it myself so, you mean tools like meshlab? or do you have reference to any other tool that might help project lidar points on image. thanks

You can try this Kitti-visualization-tool to project your lidar data to the camera image to check the correctness of your transformation matrix.

https://github.com/zzzxxxttt/simple_kitti_visualization

makaveli10, 2022-06-01:
@AOOOOOA Thanks. I was able to fix my transformations. And I am using [open-mmlab/mmdetection3d](https://github.com/open-mmlab/mmdetection3d) to use that dataset. Now, the issue is when I use ```python tools/create_data.py``` there to create kitti data the ```*.bin``` files in ```kitti_gt_database``` are all empty but I can confirm that the dataset is correct if you look at these images. 

[image]

[image]

So, I want to ask you what model/repo did you use to train on ```carla-kitti``` dataset. That would be a great help if you could share.
Thanks

AOOOOOA, 2022-06-02:
@makaveli10  I received the email about your response but cannot see on this page. It's weired.

makaveli10, 2022-06-02:
> @makaveli10 I received the email about your response but cannot see on this page. It's weired.

Weird. However, I wanted to ask about if you were successfully able to train a model on syntetic CARLA dataset(kitti format). If yes, what did you use? Seems like you used OpenPCDet with PointRCNN? How were the results?

AOOOOOA, 2022-06-02:
> Weird. However, I wanted to ask about if you were successfully able to train a model on syntetic CARLA dataset(kitti format). If yes, what did you use? Seems like you used OpenPCDet with PointRCNN? [...]

I haven't tried to train a model, but I use openpcdet to infer on my dataset. The result is normal.

makaveli10, 2022-06-02:
okay. interesting. thanks. maybe it would be a good idea then to finetune on the custom carla dataset?

Petros626, 2025-06-26:
@makaveli10 Can you provide some information how to create a own synthetic dataset in KITTI format? I think the original maps from KITTI aren't recreated, so that the city Karslruhe is simulated.