carla-simulator/carla #5455: postprocessing & quality of rgb images

knelk, 2022-05-19:
- CARLA version: CARLA_0.9.13 (downloaded ``current release`` from https://github.com/carla-simulator/carla/blob/master/Docs/download.md). 
- Platform/OS: Ubuntu 20.04.4 LTS
- Problem you have experienced: quality of images not good / no control over postprocessing
- What you expected to happen: should improve if grain jitter is disabled? need to do super-sampling?
- Steps to reproduce: render 640x480 images with ``synchronous_mode.py`` (hence using the ``.egg`` file of Carla) and check result for different maps. 

1) We noticed quite some aliasing effect around fences etc, which e.g. are transparent from some angles, but not when the car has moved a bit. Is there a Carla setting to avoid this (without manual downsampling)?

[image]

2)  In dark scenes, there is a lot of noise in the images, see attatchment. Do you know how to improve that? This is quite bad for our method.

[image]

3) We think that the noise mainly comes from Grain jitter, which is added to the image in post_processing (https://carla.readthedocs.io/en/0.9.13/ref_sensors/#basic-camera-attributes_1). Turning off post_processing completely is not an option, as images then looks super unrealistisc, e.g. no shadows, very unrealistic colors etc (see https://github.com/carla-simulator/carla/issues/173#issuecomment-362587853). How to disable grain-jitter (except with a build from source, if possible)?

4) I also noticed some ``(usually vertical) "scratch-marks"`` in the image. Do you happen to know where they come from? I saw them during day&night., here is an example during night.

[image]

jhidalgocarrio, 2022-05-19:
Yes, I also encountered the same effects. Significantly, the aliasing effect at a low image resolution.
Here is a video in Town10: https://youtu.be/oPU1XCe4pGM
I thought this issue was solved in the newest releases. Any idea @Axel1092

jhidalgocarrio, 2023-07-11:
@yyhtbs-yye It seems Carla developers have closed this issue without solving it

Could you please send an image of the case you are describing?

edelaney11, 2025-05-21:
Did you ever find a solution for this?