carla-simulator/carla #5456: License plates images getting incorrectly cropped

EricLangezaal, 2022-05-19:
I successfully followed #3429, #4537 and #4723 to get my own (partially transparant) images of license plates onto CARLA's vehicles.

Unfortunately however, the license plates seem to get cropped for some vehicles, like the volkswagen.t2_2021 for example. Below I have attached an image of what my license plate looks like, and another image of what it looks like once it is rendered on the vehicle, where you can see the cropping.

[image]
[image]

I have made sure that my license plate images are 256x256 pixels (partially transparent) like the provided texture set of license plates. Can anyone let me know how I prevent CARLA/Unreal Engine from cropping them when putting them on vehicles?

EricLangezaal, 2022-05-22:
@gsmario, @Madecu @ebadi Apologies for tagging you in here, but do any of you perhaps have an idea how I can solve this issue?

Madecu, 2022-05-23:
Hi @EricLangezaal, this is due to the UV mapping. You have two options here; Edit the UV map of the vehicle in a 3d software to match your license plate texture. Or do the other way arround, taking an already existing license plate texture in CARLA and just your texture to its shape and size in photoshop or any raster graphic editor.

EricLangezaal, 2022-05-23:
@Madecu Thank you so much! For this project I have to maintain the aspect ratio/dimensionality of my license plate images (to match real world ones), so I think that means I have to edit the UV map. Could you tell me where in Carla's Unreal Engine files I can find these UV maps? And do I need to edit it for every vehicle, or just for a single base texture?

Madecu, 2022-05-23:
You have to open Carla's Unreal project, export every vehicle mesh you want to edit the license plate, open them in a 3D software and edit de UV map, and you will probably have to edit the license plate mesh to match the aspect ratio of your texture

Madecu, 2022-05-23:
You have to edit the vehicle mesh. Open the vehicle in a 3D software, and edit the UV map of the license plate mesh inside the vehicle mesh file.
The license plate material doesn't have transparency, so If you reduce the texture it will have a black margin or the color you leave on the texture.

EricLangezaal, 2022-05-23:
Before closing this issue, I do have one final question :) (if I should start a new issue for that do let me know)

Currently the fixed normal/roughness maps in the `GenericMaterials/Licenseplates` folder do not match my license plates anymore, meaning that the depth of for example the text on the license plate is not matching the actual image used for the license plate. I can export normal/roughness maps for my pure license plate images too when creating them, but struggle getting them applied in Unreal Engine/CARLA dynamically.

For the main license plate image, there seems to be a variable in the texture at `GenericMaterials/Licenseplates/MI_LicensePlate` called `LicenseText`. Both the master material and the one in `GenericMaterials/Vehicles/` then use the output of this variable to define the rgb (and in my case alpha) of the final texture.  The `LicensePlate` script in BaseVehiclePawn uses this variable to dynamically replace the actual main license plate image from a folder of textures.

I imagine that I could do something similar for the normal image. The part I am struggling with however, is creating a variable in the `GenericMaterials/Licenseplates/MI_LicensePlate` texture that represents the normal image used. How would I do something like this? I am somewhat of a beginner with Unreal Engine, so I might be overlooking something.

closed 2022-05-24 by EricLangezaal