carla-simulator/carla #5462: CARLA 0.9.13 under WSL2 with Windows 11

daniel-bogdoll, 2022-05-23:
CARLA version: 0.9.13
Platform/OS: Ubuntu 20.04 under Windows 11 WSL2
Problem you have experienced: 
```
./CarlaUE4.sh
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
```
What you expected to happen: A UE4 windows with CARLA should open

Steps to reproduce: 

- Install Windows 11
- Install Ubuntu from the store 
[image]
- Follow this WSL2 GUI tutorial: https://docs.microsoft.com/en-us/windows/wsl/tutorials/gui-apps
- Follow this WSL2 CUDA tutorial: https://ubuntu.com/tutorials/enabling-gpu-acceleration-on-ubuntu-on-wsl2-with-the-nvidia-cuda-platform#3-install-nvidia-cuda-on-ubuntu
- Download and try to run CARLA 0.9.13

Other information (documentation you consulted, workarounds you tried):
I would love to use the newly introduce native GUI capability of WSL2 under Windows 11. Is there anything I can to do make CARLA run? Obviously, I could also just run it from Windows, but I would also test the capabilities of WSL2. I found https://github.com/carla-simulator/carla/issues/589, but it didn't help.

is there any more information necessary? :)

Avvienash, 2022-12-16:
I am facing the same problem. Did you find a fix?

lzluozheng, 2023-03-14:
@daniel-bogdoll hello! have you resolved it?i met the same trouble.

daniel-bogdoll, 2023-03-14:
With recent versions (0.9.14 etc.) I don't get the error anymore, but I'm unable to run CARLA with the GPU within WSL 2. Obviously I can launch it in Windows, but my company computer does not allow me to change any Port settings, which is why I can't connect to it from inside WSL 2.

lzluozheng, 2023-03-14:
ths!i will have a try.

Daniel-ChenJH, 2024-04-21:
I have exactly the same problem recently, with RTX3070TI LAPTOP, nvidia driver 552.12, cuda 11.5 and carla 0.9.14。Carla can run in Windows but not WSL2, with the same output. When I check by 'nvidia-smi -l 1' in WSL2, I find that there's no GPU usage from Carla. However, 'nvidia-smi' and 'nvcc -V' seems fine in my WSL2. I have tried like 'sudo prime-select nvidia' or starting Carla with '-prefernvidia' but the problem has not been solved.

> ./CarlaUE4.sh
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744
Malloc Size=131160 LargeMemoryPoolOffset=327928
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault

Daniel-ChenJH, 2024-04-21:
Is the problem caused by port forwarding or firewall of Windows? It seems that WSL2 needs to connect to 127.0.0.1 by the localhost on Windows.

eyabesbes, 2024-04-22:
> CARLA version: 0.9.13 Platform/OS: Ubuntu 20.04 under Windows 11 WSL2 Problem you have experienced:
>
> ```
> ./CarlaUE4.sh
> 4.26.2-0+++UE4+Release-4.26 522 0
> Disabling core dumps.
> Signal 11 caught. [...]

Hi,
I have also been facing the same error. I have already Carla installed on Windows but when I install carla on WSL2 I get the same error ? did you manage to resolve it ?