carla-simulator/carla #5467: ERROR: buildNavigation: Could not create solid heightfield.

javiercalle97, 2022-05-24:
I'm using Carla 0.9.13 and I'm trying to import a custom map created with roadrunner. 

I have followed the steps described in this tutorial https://carla.readthedocs.io/en/0.9.13/tuto_M_add_map_source/ (using the make import method). But I'm facing some problems.

The main problem is that I can't generate the pedestrian navegation file, the pedestrian can spawn  using `world.spawn_actor()` but they don't move (I have followed this tutorial https://carla.readthedocs.io/en/0.9.13/tuto_M_generate_pedestrian_navigation/). 

When I try to create the .bin file i found this error: 
ERROR: buildNavigation: Could not create solid heightfield. 

I have found this issue in the github repo: https://github.com/carla-simulator/carla/issues/5106 
but nothing more. Any help is welcomed. 

Thanks in advance!

MMNavetty, 2022-06-22:
same problem here, does anyone has a solution ?

MMNavetty, 2022-06-28:
For those with these issue, try to export only the mesh related to pedestrian navigation into the .obj file. It resolves my error.

javiercalle97, 2022-06-29:
Exporting only the meshes related to the pedestrian navigation has solved the problem. Thanks!

closed 2022-06-29 by javiercalle97

devzdevelop, 2023-06-11:
> Hi All, the map that i have do not include pedestrian navigation, its a simple map with 4 way protected traffic signal exported from Road runner, i still get this error. Truly appreciate your suggestion. How do we export mesh related to specific concern into .obj? Kindly let me know -Rani

I have the same issue, did you find a way to resolved it?