carla-simulator/carla #5469: The C Compiler "...\cl.exe" is not able to compile a simple test programm on "make PythonAPI" (Windows 11)

AuberDjine, 2022-05-24:
Hey People,

Im trying to Build Carla, but when I try and type "make PythonAPI" I get the Error displayed in the Screenshot:

[image]

Now I guess its a Problem with the Visual Studio Compiler. However I dont know what to do as I already tried reinstalling Visual Studio...

Does anyone know what I could try to Fix the Issue?

Reyen994, 2022-05-25:
Carla doesn't work on Windows11. I had the same Problem. If i used Win10 it works.

sp2150, 2022-05-30:
I do have the same issue ... 

Any news how to fix it using Windows 11 ?

PatroSV19, 2022-07-12:
I believe that I came across the same error.

This thread seems to have had the same issue and came up with a solution: #5351

Blyron, 2024-04-12:
Are you ok with closing this issue as you have found a solution?

closed 2024-05-17 by MarcelPiNacy-CVC