carla-simulator/carla #5482: pip install carla failing

MachiriMarodi, 2022-06-01:
Hi, I have been trying to complete the installation process for sometime now. I keep running into this error when I pip install carla. Does anyone have any ideas around how to resolve this?

I have python 3.9 installed on my machine, I am using an Windows OS.

C:\Users\machi>pip install carla
Collecting carla
  Using cached carla-0.9.5.tar.gz (3.9 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\machi\AppData\Local\Temp\pip-install-em_3m8q4\carla_4953a80ecd3343809248a969a7c4c1f9\setup.py", line 114, in <module>
          ext_modules=get_libcarla_extensions(),
        File "C:\Users\machi\AppData\Local\Temp\pip-install-em_3m8q4\carla_4953a80ecd3343809248a969a7c4c1f9\setup.py", line 75, in get_libcarla_extensions
          libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
      FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

sreenivaskrishnanair, 2022-10-02:
failing for me too

pip3 install carla
Collecting carla
  Using cached carla-0.9.5.tar.gz (3.9 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\sreen\AppData\Local\Temp\pip-install-5g70b6jq\carla_b0fc1119f986485a9f6f7e424b78af99\setup.py", line 114, in <module>
          ext_modules=get_libcarla_extensions(),
        File "C:\Users\sreen\AppData\Local\Temp\pip-install-5g70b6jq\carla_b0fc1119f986485a9f6f7e424b78af99\setup.py", line 75, in get_libcarla_extensions
          libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
      FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

Yosefnoorani, 2022-10-02:
Me too :(

Any solution?

sreenivaskrishnanair, 2022-10-02:
I was able to get it running :) 

from what I understand CARLA can only work with python3.7, I was running on python 3.10.

Downgrading python to 3.7 fixed my issues.

Yosefnoorani, 2022-10-03:
Thank you!
It is solved!

c-s-westphal, 2022-11-15:
Downgrading to 3.7 gave me the same error, any ideas?

Kin-Zhang, 2023-02-17:
If carla cannot be installed, it is recommended to use the whl installation solution:

1. Download the wheel from here: [https://pypi.org/project/carla/#files](https://pypi.org/project/carla/#files)
2. Unzip and download as shown in Figure ① below
3. Move to `${HOME}$/.local/lib/python3.8/site-packages` as shown below ②
4. Finally, when grep carla below, as shown in figure ③ below

[image]

MJy1023, 2023-08-28:
Same problem. 

Defaulting to user installation because normal site-packages is not writeable
Collecting carla
  Using cached carla-0.9.5.tar.gz (3.9 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-_w7w01nj/carla_09d577e291614b99b70f02f9b3c3d74c/setup.py", line 114, in <module>
          ext_modules=get_libcarla_extensions(),
        File "/tmp/pip-install-_w7w01nj/carla_09d577e291614b99b70f02f9b3c3d74c/setup.py", line 31, in get_libcarla_extensions
          linux_distro = platform.dist()[0]  # pylint: disable=W1505
      AttributeError: module 'platform' has no attribute 'dist'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

CupCodeIr, 2024-06-03:
> Same problem.
> 
> Defaulting to user installation because normal site-packages is not writeable Collecting carla Using cached carla-0.9.5.tar.gz (3.9 kB) Preparing metadata (setup.py) ... error error: subprocess-exited-with-error
> 
> × python setup.py egg_info did not run successfully. │ exit code: 1 ╰─> [8 lines of output] Traceback (most recent call last): File "", line 2, in File "", line 34, in File "/tmp/pip-install-_w7w01nj/carla_09d577e291614b99b70f02f9b3c3d74c/setup.py", line 114, in ext_modules=get_libcarla_extensions(), File "/tmp/pip-install-_w7w01nj/carla_09d577e291614b99b70f02f9b3c3d74c/setup.py", line 31, in get_libcarla_extensions linux_distro = platform.dist()[0] # pylint: disable=W1505 AttributeError: module 'platform' has no attribute 'dist' [end of output]
> 
> note: This error originates from a subprocess, and is likely not a problem with pip. error: metadata-generation-failed
> 
> × Encountered error while generating package metadata. ╰─> See above for output.
> 
> note: This is an issue with the package mentioned above, not pip. hint: See above for details.

Same error