carla-simulator/carla #5484: Carla support for Autoware.auto

zyan1234, 2022-06-01:
Hi,
I am keen to know how carla works with Autoware.I went through the documentation , got to know that it does support various versions of autoware.ai.
But I wanted to know if Carla supports Autoware.Auto
Does Carla supports Autoware.auto.Please help.
Thank you.

Sam827-r, 2022-06-22:
@zyan1234 did you got any references. mine question also same.
Does Carla supports AutowareAuto??

Blyron, 2026-11-12:
Hi @zyan1234 (and @Sam827-r),

Sorry for the very late reply. CARLA never officially supported **Autoware.Auto**. The Autoware Foundation has since deprecated Autoware.Auto and folded it into what is now simply **Autoware** (Autoware Core/Universe, ROS 2). So if you are starting today, target Autoware (Universe), not Autoware.Auto.

**On CARLA 0.9.x:** the Autoware project maintains [`autoware_carla_interface`](https://github.com/autowarefoundation/autoware_universe/tree/main/simulator/autoware_carla_interface) in Autoware Universe. It connects Autoware to **CARLA 0.9.15** on Ubuntu 22.04 / ROS 2 Humble through the Python API. Its README covers installation, the maps to download and the sensor setup.

**On `ue58-dev` / CARLA 1.0:** we are migrating to the `ue58-dev` branch (Unreal Engine 5.8), which will become CARLA 1.0. There, Autoware is supported natively, with **no bridge process**. The server publishes sensor data and vehicle status straight onto ROS 2 (`/vehicle/status/*`) and applies Autoware's control commands (`/control/command/*`) to the ego vehicle. Two modes ship with it:
- `classical`: the full Autoware stack (NDT localization on a point-cloud map, lidar perception, planning, trajectory follower).
- `e2e`: the camera-only VAD end-to-end model.

Everything is in `PythonAPI/examples/av_stacks/autoware/`. That folder has an installer for Autoware, a launcher that starts CARLA and Autoware together, and the tools that generate the lanelet2 and point-cloud maps Autoware needs for each town. Docs:
- Autoware on CARLA: https://carla-ue5.readthedocs.io/en/latest/autoware/
- Native ROS 2 interface: https://carla-ue5.readthedocs.io/en/latest/ros2_native/

Requirements on ue58-dev:
- CARLA built with `-DENABLE_ROS2=ON`. It is off by default and cannot be turned on after the build.
- The server started with the `-ros2` flag.
- Ubuntu 22.04 + ROS 2 Humble, or Ubuntu 24.04 + ROS 2 Jazzy.
- Raise the UDP socket buffers to 64 MB, as the docs describe. Otherwise the lidar stream overruns and Autoware emergency-stops the vehicle.

There is also an agent skill that walks through the whole setup: [`Skills/ue58/run-autoware-ue58`](https://github.com/carla-simulator/carla/tree/ue58-dev/Skills/ue58/run-autoware-ue58) (see https://carla-ue5.readthedocs.io/en/latest/agent_skills/).

closed 2026-11-12 by Blyron