carla-simulator/carla #5490: Add ring and time data to every lidar point

PeteBai, 2022-06-03:
Currently the simulated lidar sensor outputs x, y, z and intensity, but modern lidar sensors often include ring and time data for every detection point. How can we add such feature to carla/pythonAPI/ros-bridge?

fawadshahid, 2022-07-21:
look here! https://github.com/carla-simulator/ros-bridge/issues/416

fgoudreault, 2022-07-26:
In the LiDAR implementation, all rays are cast at the same time during a frame. Thus the timestamp could be the frame ID times the delta timeout between frames. I am not sure what the ring is but if it is the channel number (i.e.: the laser idx), then using the regular lidar it might be hard to get. Using the semantic lidar which returns all the points (i.e.: no points dropped due to noise), you can reshape the pts array (with shape npts x nfeatures) to (nchannels x npts_per_channel x nfeatures.

Hopes this helps!

Robotics010, 2023-04-08:
> look here! [carla-simulator/ros-bridge#416](https://github.com/carla-simulator/ros-bridge/issues/416)

It looks like we are waiting for such a feature from carla developers (not from carla-ros-bridge) side but there are no news since [Mar 1, 2021](https://github.com/carla-simulator/ros-bridge/pull/431#issuecomment-788088970)