carla-simulator/carla #5498: Unclear interaction between carla executable -fps, -benchmark arguments and fixed-timestep synchronous mode

Pyrestone, 2022-06-07:
Hi CARLA-Team,

I have a question regarding the interactions between a synchronous mode fixed-timestep simulation and the arguments provided to the carla executable.

My goal is to run a completely repeatable (deterministic) simulation, therefore I'm trying to figure out which arguments and options affect this determinism.

As you probably know, the `./CarlaUE4.sh` script takes various arguments.
Intuitively, I would expect the `-fps` and `-benchmark`to no longer have an effect on determinism if i set synchronous mode to true.
However, at least with my (deterministic, synchronous) traffic manager instance, I can only get the same traffic when I set these args the same.

Could you explain how the different args (-benchmark, -fps) interact with a fixed-delta synchronous simulation?
Does -benchmark override -fps? does the fixed_delta_seconds override fps?
The documentation does not explain these args well enough.

Also, (unrelated but important for determinism) does resetting traffic lights achieve the same as reloading the world after setting synchronous mode? I found this to (mostly) be the case, but couldn't find anything in the docs.

tk166, 2022-06-13:
Same concern. I personally don't use -fps and -benchmark arguments any more.

Pyrestone, 2024-02-04:
still no response. Don't close this just because you take more than a year to respond.