carla-simulator/carla #5502: No-rendering doesn't stop producing camera data

SoMuchSerenity, 2022-06-10:
CARLA version: 0.9.12
Platform/OS: Windows 11
Problem you have experienced: No rendering option still outputs camera images
What you expected to happen: No rendering option should return no camera results

Steps to reproduce: 
I attached a RGB camera to my car and I have tried 
`
settings = world.get_settings()
settings.no_rendering_mode = True
world.apply_settings(settings)
`
and also the scripts `python config.py --no-rendering` to disable the rendering mode, yet still got image output in my designated directory. Just wonder did I do something wrong or is it a possible bug?

brscholz, 2022-06-28:
The no rendering mode suppresses the generation of the spectator render target. Spawning a camera creates an extra render target for that camera, that's why you still get output images. But why would you add a camera in no rendering mode?

SoMuchSerenity, 2022-07-01:
@brscholz. Hi Scholz, the reason I added a camera in no rendering mode is that according to this tutorial: https://carla.readthedocs.io/en/0.9.12/adv_rendering_options/ , any camera or GPU sensor should return empty data. It also says UE engine would skip everything regarding graphics, so I was experimenting with it.

Morphlng, 2023-08-16:
> The no rendering mode suppresses the generation of the spectator render target.

According to the doc, any GPU related computation should be skipped? If this only disabled spectator view, off-screen rendering might be a better solution?

TzabarDolev, 2023-10-04:
Hi, I wrote a blog post about this issue on the CarlaSimBlog, including a code tutorial for off screen rendering.
https://carlasimblog.wordpress.com/2023/10/04/rendering-images-off-screen-in-the-carla-simulator/

the-big-bad-wolf, 2024-09-19:
The docs say that GPU-based sensors won't work in no-rendering mode, but I am not sure which sensors are GPU-based. I was (perhaps way too optimistically) hoping that the semantic LIDAR would work in no-rendering mode since it is not listed as a camera-type sensor, but I was not able to verify this due to rendering being enabled when needed. Is there a list of sensors which don't require any rendering?

Blyron, 2026-11-12:
This is intended: no-rendering mode only switches off the spectator view. Camera sensors keep rendering, which is how multi-GPU secondaries work. See the no-rendering section of [Rendering](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_rendering.md). Do not spawn cameras if you want no GPU work. Closing.

closed 2026-11-12 by Blyron