carla-simulator/carla #5503: ./CarlaUE4.sh 4.26.2-0+++UE4+Release-4.26 522 0 Disabling core dumps. I have this issue and when the window CarlaUE4 shows just the see. what should I do for the next step?

ElishBanani, 2022-06-10:


GittUser2021, 2023-05-05:
having the same issue, any solution to this ?

MarcelPiNacy-CVC, 2024-05-17:
Hi,

I'm closing this issue.
Feel free to reopen it, but please provide more info.

Marcel

closed 2024-05-17 by MarcelPiNacy-CVC

ak-maker, 2024-09-10:
Hi! I have met the same issue on windows11, WSL2, ubuntu22.04. I have followed the linux package installation instruction on installing carla-0.9.15 from https://carla.readthedocs.io/en/latest/start_quickstart/.  
[image]

MarcelPiNacy-CVC, 2024-09-12:
Hi @ak-maker,

Could you share the output of the vulkaninfo and nvidia-smi commands?

Revathy2025, 2025-07-31:
did anyone solved this problem.. kindly tell solution for this

lzy886, 2025-12-03:
I encounter the same problem on Ubuntu