carla-simulator/carla #5504: Odd positioning when importing map as xodr

ar4development, 2022-06-11:
CARLA version: 0.9.13
Platform/OS: Linux Ubuntu 20.04

Hi community,

I'm developing OpenDrive editor and at some point of development my maps became valid for some basic scenarios (like having no junctions). I tried to import maps from my editor to Carla and noticed the issue.

The issue is that while the geometry of the roads looks fine, the actual positioning of the roads is wrong. I prepared minimal reproducible example which is the following:

Say we have a very trivial map. **As trivial as it ever possible:**

```
<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<OpenDRIVE>
    <header date="2022-06-11T11:47:55.135177" east="40" name="" north="10" revMajor="1" revMinor="7" south="6" vendor="Alexey Razgulyaev" version="1" west="0">
        <geoReference>
            <![CDATA[+proj=tmerc +lat_0=0 +lon_0=0 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]>
        </geoReference>
    </header>
    <road id="2" junction="-1" length="40" name="Road_2">
        <planView>
            <geometry hdg="0.0" length="40" s="0.0" x="0" y="10">
                <line/>
            </geometry>
        </planView>
        <lanes>
            <laneSection s="0.0">
                <left/>
                <center>
                    <lane id="0" level="false" type="none">
                        <roadMark laneChange="none" material="standard" sOffset="0.0" type="none" width="0.15"/>
                    </lane>
                </center>
                <right>
                    <lane id="-1" level="false" type="driving">
                        <width a="4" b="0.0" c="0.0" d="0.0" sOffset="0.0"/>
                        <roadMark laneChange="none" material="standard" sOffset="0.0" type="none" width="0.15"/>
                    </lane>
                </right>
            </laneSection>
        </lanes>
    </road>
</OpenDRIVE>
```
This map basically consists of a **single straight road** that **has no any heading rotation**, starts at {0, 10} and has the length of 40 meters. It also has a single lane width width = 4 m.

I also have python script that draws x and y axes and where the road has to be according its positioning on the map:

```
import time
import carla
import logging

def main():
    client = carla.Client('127.0.0.1', 2000)
    client.set_timeout(10.0)

    try:
        world = client.get_world()
        world.debug.draw_line(carla.Location(0, 0, 0), carla.Location(100, 0, 0), 0.1, carla.Color(0, 255, 0), 0)
        world.debug.draw_line(carla.Location(0, 0, 0), carla.Location(0, 100, 0), 0.1, carla.Color(255, 0, 0), 0)

        world.debug.draw_line(carla.Location(0, 10, 0), carla.Location(40, 10, 0), 0.1, carla.Color(0, 0, 255), 0)
        world.debug.draw_line(carla.Location(40, 10, 0), carla.Location(40, 6, 0), 0.1, carla.Color(0, 0, 255), 0)
        world.debug.draw_line(carla.Location(40, 6, 0), carla.Location(0, 6, 0), 0.1, carla.Color(0, 0, 255), 0)
        world.debug.draw_line(carla.Location(0, 6, 0), carla.Location(0, 10, 0), 0.1, carla.Color(0, 0, 255), 0)
    finally:

        time.sleep(0.5)

if __name__ == '__main__':

    try:
        main()
    except KeyboardInterrupt:
        pass
    finally:
        print('\ndone.')
```
That script draws X green axis and Y red axis heading to their positive directions and blue expected boundary of the road.

**OBSERVED RESULT**

However when I run everything (run Carla, import my map and run my script) I get this result:

[image]

It is clearly seen that the road is positioned not where it is assumed to be. Appreciate your help.

fangchaooo, 2022-06-14:
I think your problem has some realtion with me. look https://github.com/carla-simulator/carla/issues/5510. Two map have same (x,y)->(lat,long), but road load direction error. I want to find this issue

Blyron, 2026-11-12:
This is by design: OpenDRIVE is right-handed and CARLA (Unreal Engine) is left-handed, so the parser negates y. A point (x, y) in OpenDRIVE appears at (x, -y) in CARLA. See [Coordinate conventions](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/core_coordinate_conventions.md). Closing.

closed 2026-11-12 by Blyron