carla-simulator/carla #5506: How to collect the sensor data from multiple cars

RostyslavUA, 2022-06-12:
I want to collect the data from the sensors located on multiple cars in CARLA 9.13. I run the simulation in the synchronous mode. The problem happens, when I try to listen to multiple sensors i.e.
```
def slidar_callback(sen_id):
    print(sen_id)
```
```
for sen in world.get_actors().filter('sensor.*'):
    sen.listen(lambda pcl_data: slidar_callback(sen.id))
```
In this case the id of only one (out of 15) sensor gets returned. As a result, the data from only one sensor is being saved.
What am I doing wrong?

UVCSzb, 2023-05-05:
Did you resolve the problem?

AlaaBenZekri, 2023-11-16:
This may be a bit late but it would be helpful for anyone else facing a similar issue; if you are defining spawning multiple sensors and the callback in their listen methods is using parameters having the same name (even though the value may have changed between the spawning of the different sensors), only the last definition of those parameters would be taken into consideration and it would be having the same value for different sensors.
This would be the same if you are trying to collect data using multiple vehicles where the vehicles (and/or their sensors) are defined in a loop.

One way to avoid this issue is to follow the way the sensors were defined in [this script](https://github.com/carla-simulator/carla/blob/master/PythonAPI/test/smoke/test_sync.py).

Blyron, 2026-11-12:
This is Python's late binding of loop variables in the lambda: every callback reads `sen` after the loop has finished. Bind the value when creating the lambda, for example `sensor.listen(lambda data, sid=sen.id: callback(sid, data))`. Closing.

closed 2026-11-12 by Blyron