carla-simulator/carla #5507: RuntimeError: time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

ShubhamAbhayDeshpande, 2022-06-12:


ShubhamAbhayDeshpande, 2022-06-13:
Hello, 

I am new to Carla. 

Re-posting this issue because earlier similar issues are closed without any concrete solution.

I am trying to run carla on Ubuntu 22.04. I have tried multiple versions of carla  (0.9.6, 0.9.4 etc.) 

The complete error is as follows:

Traceback (most recent call last):
  File "automatic_control.py", line 830, in <module>
    main()
  File "automatic_control.py", line 823, in main
    game_loop(args)
  File "automatic_control.py", line 738, in game_loop
    world = World(client.get_world(), hud, args.filter)
RuntimeError: time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

The configuration of my computer is as follows
RAM 32 GB.
Graphics card 2 GB NVIDIA GTX 1650. 

Please, keep the threads focused and single-themed. Make them easy to find
for everyone.

Pyrestone, 2022-06-14:
Did you start the simulator as well?
the carla python API connects to an existing server, so you need to start that server before you run your python code.
See https://carla.readthedocs.io/en/latest/start_quickstart/#running-carla

ar4development, 2022-06-14:
Looks like you have not enough of gpu ram. Usually one observes such behavior when the hardware does not let Carla to reply fast.

ShubhamAbhayDeshpande, 2022-06-14:
> Did you start the simulator as well? the carla python API connects to an existing server, so you need to start that server before you run your python code. See https://carla.readthedocs.io/en/latest/start_quickstart/#running-carla

Hello, 

Thanks for the solution. It worked. 

But I have to use python 2.7 to run the PythonAPIs for carla 0.9.4. Any idea why it is like that?

closed 2022-06-26 by ShubhamAbhayDeshpande

whitehatjunior35, 2023-01-16:
Hi, I tried changing the port number to 3000, but throws an error like this 
Traceback (most recent call last): File "[project-215.py](https://project-215.py/)", line 56, in <module> 
world = client.get_world() 
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to [127.0.0.1:3000](https://127.0.0.1:3000/)

SushilaRatre, 2023-08-02:
Hi, Initially I was also facing issue related to the following:

C:\CARLA_0.9.5\PythonAPI\examples>py -3.7 test1.py
destroying actors
done.
Traceback (most recent call last):
  File "test1.py", line 39, in <module>
    world = client.get_world()
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to localhost:3000

Now, I am able to solve the issue. 
Firstly, I have started my server with the following command:
CarlaUE4.exe -windowed -caral-server -benchmark-fps=20
[image]
It has started my server.

[image]

Now I have run the client machine with the following code:
C:\CARLA_0.9.5\PythonAPI\examples>py -3.7 test1.py

Client as an object in the test1.py
[image]

And finally, I am getting the output.
[image]

So finally, I have established the connection between client and server.

SushilaRatre, 2023-08-02:
> Hello,
> 
> I am new to Carla.
> 
> Re-posting this issue because earlier similar issues are closed without any concrete solution.
> 
> I am trying to run carla on Ubuntu 22.04. I have tried multiple versions of carla (0.9.6, 0.9.4 etc.)
> 
> The complete error is as follows:
> 
> Traceback (most recent call last): File "automatic_control.py", line 830, in main() File "automatic_control.py", line 823, in main game_loop(args) File "automatic_control.py", line 738, in game_loop world = World(client.get_world(), hud, args.filter) RuntimeError: time-out of 20000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000
> 
> The configuration of my computer is as follows RAM 32 GB. Graphics card 2 GB NVIDIA GTX 1650.
> 
> Please, keep the threads focused and single-themed. Make them easy to find for everyone.

I have solved the issue #5507. Please check.

aditya324jhj, 2024-01-02:
Hello!
I have tried all the above recommendations but it is still not working.
I have 24 gb of ddr4 ram ,gtx1650 ti, and a ryzen 7 .
When I try to run it using this command it works -->CarlaUE4.exe -windowed -caral-server -benchmark-fps=20
but else it doesn't work.
Thank you....

SushilaRatre, 2024-01-04:
> Hello! I have tried all the above recommendations but it is still not working. I have 24 gb of ddr4 ram ,gtx1650 ti, and a ryzen 7 . When I try to run it using this command it works -->CarlaUE4.exe -windowed -caral-server -benchmark-fps=20 but else it doesn't work. Thank you....

Hello Aditya,

Please check your Carla version. It should be 9.5. older version.
C:\CARLA_0.9.5\PythonAPI\

aditya324jhj, 2024-03-07:
> Hello Aditya,
>
> Please check your Carla version. It should be 9.5. older version. C:\CARLA_0.9.5\PythonAPI\

Thanks for the relpy,
I am already on version 9.15 but it doesn't seem to work...
I hope you can help because my final project is on carla simulator which I have to submit by April :( .....

SushilaRatre, 2024-03-07:
> > > Hello! I have tried all the above recommendations but it is still not working. I have 24 gb of ddr4 ram ,gtx1650 ti, and a ryzen 7 . When I try to run it using this command it works -->CarlaUE4.exe -windowed -caral-server -benchmark-fps=20 but else it doesn't work. Thank you....
> > 
> > 
> > Hello Aditya,
> > Please check your Carla version. It should be 9.5. older version. C:\CARLA_0.9.5\PythonAPI\
> 
> Thanks for the relpy, I am already on version 9.15 but it doesn't seem to work... I hope you can help because my semester's final project is on carla simulator which I have to submit by April :( .....

Congratulation Aditya!!!,

You have successfully run the server.
The second step is to establish a connection between the server and the client.

Now, run the client machine with the following code in another terminal/command prompt.
C:\CARLA_0.9.5\PythonAPI\examples>py -3.7 test1.py

After establishing the connection between client and server, you can run your own code. Please refer tutorial for coding that is already given on the Carla website.

aditya324jhj, 2024-03-07:
> Congratulation Aditya!!!,
>
> You have successfully run the server. The second step is to establish a connection between the server and the client.
>
> Now, run the client machine with the following code [...]

Brother I don't know if you are not able to understand what I meant....
"test1.py" file doesn't work for me still because of timeout error.....

SushilaRatre, 2024-03-07:
> Brother I don't know if you are not able to understand what I meant.... "test1.py" file doesn't work for me still because of timeout error.....

Brother, I believe that you have successfully run your server. And you are not able to run Python code. I would suggest you that you should try CARLA_0.9.5 and run the Python code. Sometimes in the higher version and latest version of CARLA will give errors due to some missing files and system compatibility.

Initially, I had tried for 9.13, 9.12, 9.11 , but I was facing lots of issues to run CARLA. So I have tried for CARLA_0.9.5.

aditya324jhj, 2024-03-07:
Ran it in 0.9.5 :
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

aditya324jhj, 2024-03-07:
Thanks to all, finally it is working the only recommendation I have is that use CARLA_0.9.5 and use older version of python like 2.8. That's it....
Thanks once again.....