carla-simulator/carla #5518: What does the value _dot in the controller.py means when the ego_vehicle is at the waypoint?

holymoly-JihyeSeo, 2022-06-16:
Hi, im trying to study the Carla code source to understand how the controller for steering works when using Waypoint. 

So, at the controller.py located carla\PythonAPI\carla\agents\navigation\controller.py  
class PIDLateralController() is defined as followed:

```
class PIDLateralController():
    """
    PIDLateralController implements lateral control using a PID.
    """

    def __init__(self, vehicle, offset=0, K_P=1.0, K_I=0.0, K_D=0.0, dt=0.03):
        """
        Constructor method.
            :param vehicle: actor to apply to local planner logic onto
            :param offset: distance to the center line. If might cause issues if the value
                is large enough to make the vehicle invade other lanes.
            :param K_P: Proportional term
            :param K_D: Differential term
            :param K_I: Integral term
            :param dt: time differential in seconds
        """
        self._vehicle = vehicle
        self._k_p = K_P
        self._k_i = K_I
        self._k_d = K_D
        self._dt = dt
        self._offset = offset
        self._e_buffer = deque(maxlen=10)

    def run_step(self, waypoint):
        """
        Execute one step of lateral control to steer
        the vehicle towards a certain waypoin.
            :param waypoint: target waypoint
            :return: steering control in the range [-1, 1] where:
            -1 maximum steering to left
            +1 maximum steering to right
        """
        return self._pid_control(waypoint, self._vehicle.get_transform())

    def _pid_control(self, waypoint, vehicle_transform):
        """
        Estimate the steering angle of the vehicle based on the PID equations
            :param waypoint: target waypoint
            :param vehicle_transform: current transform of the vehicle
            :return: steering control in the range [-1, 1]
        """
        # Get the ego's location and forward vector
        ego_loc = vehicle_transform.location
        v_vec = vehicle_transform.get_forward_vector()
        v_vec = np.array([v_vec.x, v_vec.y, 0.0])

        # Get the vector vehicle-target_wp
        if self._offset != 0:
            # Displace the wp to the side
            w_tran = waypoint.transform
            r_vec = w_tran.get_right_vector()
            w_loc = w_tran.location + carla.Location(x=self._offset*r_vec.x,
                                                         y=self._offset*r_vec.y)
        else:
            w_loc = waypoint.transform.location

        w_vec = np.array([w_loc.x - ego_loc.x,
                          w_loc.y - ego_loc.y,
                          0.0])

        wv_linalg = np.linalg.norm(w_vec) * np.linalg.norm(v_vec)
        if wv_linalg == 0:
            _dot = 1
        else:
            _dot = math.acos(np.clip(np.dot(w_vec, v_vec) / (wv_linalg), -1.0, 1.0))
        _cross = np.cross(v_vec, w_vec)
        if _cross[2] < 0:
            _dot *= -1.0

        self._e_buffer.append(_dot)
        if len(self._e_buffer) >= 2:
            _de = (self._e_buffer[-1] - self._e_buffer[-2]) / self._dt
            _ie = sum(self._e_buffer) * self._dt
        else:
            _de = 0.0
            _ie = 0.0

        return np.clip((self._k_p * _dot) + (self._k_d * _de) + (self._k_i * _ie), -1.0, 1.0)

    def change_parameters(self, K_P, K_I, K_D, dt):
        """Changes the PID parameters"""
        self._k_p = K_P
        self._k_i = K_I
        self._k_d = K_D
        self._dt = dt
```

with this, i understood as when the ego vehicle's location is at the waypoint( ego's location = waypoint.transform), wv_linalg == 0 and thus _dot =1 

and the return value for function _pid_control is 1.
finally, as current_steering = self._lat_controller.run_step is 1 which means maximum steering to right.

However, it does not seem to be correct as that means ego_vehicle steer right to the maximum when it is at waypoint.
as expected, isn't _dot should be 0 so that the ego vehicle does not move?

holymoly-JihyeSeo, 2022-06-16:
The  python file link is here :

[PythonAPI/carla/agents/navigation/controller.py]
(https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/controller.py)

ar4development, 2022-06-16:
Looks like `_dot` is radians between heading of `ego` and vector from `ego` to `target`. If so it would be more natural to have it set to zero indeed. 

However I am not sure if it really matters because the method "estimates" the steering which is valid only when you're not at the endpoint. Probably it is assumed that you should not call this method when you currently at the target.

M-Colley, 2026-10-26:
Draft PR #10001 proposes the `ue58-dev` PID edge-case fix for coincident target/vehicle positions, where the direction vector has zero length. It includes offline controller regressions. Live driving behavior remains unrun.

Blyron, 2026-11-12:
In `PIDLateralController`, `_dot` is the heading error angle: `acos` of the normalised dot product between the vehicle's forward vector and the vector to the target waypoint, with its sign taken from the z component of their cross product (negative when the target is to the left). The PID then acts on that signed angle. Closing.

closed 2026-11-12 by Blyron