carla-simulator/carla #5520: Any idea how to add checkboard control code for PythonAPI use?

xiao6768, 2022-06-16:
### Discussed in https://github.com/carla-simulator/carla/discussions/5519

<div type='discussions-op-text'>

<sup>Originally posted by **xiao6768** June 16, 2022</sup>
Hi there,

My purpose is to add checkboard into Carla so that 3rd-party open source AD solution Autoware.AI may use checkboard for Lidar && Vision calibration.
Firstly ,there is a similar request https://github.com/carla-simulator/carla/issues/2696  but not complete yet. I have to do by myself.
At this point I had build mesh-components , by putting into Town03 map,  vision and lidar detection as expected shown as below.
[image]

Now I want to control them by carla-python-client.  like "SpawnActor(checbboard_blueprint, transform)", 
Then I do:
1. change checkboard component's parent class to CarlaActorBluePrintFactory
2.  Implement "Make Board Denfinition" in *.h file

BTW, my envirement is ubuntu18.04, source code build. UE4 : 4.24.3  carla: 0.9.10
By recompile all (make launch , then play),

Use following code , I EXPECTED checkboard will be appear on blueprint_library.  BUT NOT.
"
import carla
client = carla.Client('127.0.0.1', 2000)
client.set_timeout(3)
world = client.get_world()
bp_library = world.get_blueprint_library()
print(bp_library)
"

Any suggestion would be great appreciated!!!!
thanks.

</div>

xiao6768, 2022-06-20:
Good news! This issue has been resolved . We are planning to merge into main branch, need approve from company. 
Any one please feel free to contact me if you have any questions.

shr1997, 2022-07-01:
> 

I've been looking for it for a long time. This is exactly what I need. Is it convenient to share the checkboard map or code? Thank you very much

AbanobSoliman, 2022-07-01:
Hello everyone, 
As part of our recent [research work](https://arxiv.org/abs/2206.13455), we needed to have exact (calibrated) values for all CARLA cameras' intrinsic parameters (RGB, Depth, DVS). Accordingly, we added calibration targets (Checkerboard and AprilGrid) to the Town3 map, and calibrated RGB, and DVS cameras using SOTA algorithms. Check out the repo here: [IBISCape](https://github.com/AbanobSoliman/IBISCape).
Thanks.

shr1997, 2022-07-04:
‘Then, open the CARLA-Unreal Engine interface and press Compile, Build and finally Play.’
I'm really sorry, my Carla is installed under ubuntu20, and I can't find where to open the&nbsp;CARLA-Unreal Engine interface，I tried to replace the map file directly, but failed
I want to know how to import the map files

xiao6768, 2022-07-04:
Firstly, My compile Env. is ubuntu18.04, carla 0.9.10, and UE4 is 4.24.3.
You have to following[ this instruction](https://carla.readthedocs.io/en/0.9.10/build_linux/) get UE4 source code compile OK, then compile Carla source code.