carla-simulator/carla #5537: Roadmap for Unreal Engine 5?

sudalsol, 2022-06-24:
Hello CARLA team, thank you for your hard work on this simulator!

With UE5 released, are there any plans to upgrade to UE 5? Thank you!

rec5885, 2022-06-30:
I wonder this answer too , ue5 is really fast , if we can get on to ue5 , simulation quality and scale increase by order of magnitude

ARtSSt, 2022-07-18:
+2

halittefe, 2022-07-19:
+111111

gawinwong, 2022-09-02:
UE5 is great, what concerns me more is that when will 4.27 be supported

junaidbinsarfraz, 2022-11-17:
Still looking for a road map for UE5. It is quite important for us.

eliabruni, 2023-01-28:
> UE5 is great, what concerns me more is that when will 4.27 be supported

Teaser: At [Saivvy](https://github.com/saivvy) we are about to release 4.27 support for CARLA - stay tuned!

olegprog, 2023-04-20:
+1 to the question, 
Dear CARLA team, are there any plans or  news related to upgrading to UE5?

asetsuna, 2023-05-12:
> > UE5 is great, what concerns me more is that when will 4.27 be supported
> 
> Teaser: At [Saivvy](https://github.com/saivvy) we are about to release 4.27 support for CARLA - stay tuned!

@eliabruni That's great work. Is there any time for the releasing ?

Blyron, 2024-05-03:
Hello! We have active UE5.3 Development branch at ue5-dev. You will have news soon I will be closing this issue