carla-simulator/carla #5538: Making Carla build with Python 3.10

itivds, 2022-06-24:
Hi all, 

we want to use the Carla ROS Bridge with the recently released version of ROS2, Humble. However, Humble is targeting Ubuntu 22, which is delivered with Python 3.10 as default. Thus, it would be great if the compiling process of Carla would support using Python 3.10 for building the PythonAPI. If anyone has the same struggle, the following steps led to a successful build of PythonAPI, make launch and make package to get a new release of Carla with python 3.10 support:
- Build on Ubuntu 20 (not sure if this is necessary)
- Bump Boost to 1.75. For that, changes in the BoostFiles patches were required to enable the no_exception build. See attached files, which are based on the official lib files [Rational.hpp](https://www.boost.org/doc/libs/1_75_0/boost/rational.hpp) and [Read.hpp](https://www.boost.org/doc/libs/1_75_0/boost/geometry/io/wkt/read.hpp) What had to be changed is, to add `# include <boost/detail/no_exceptions_support.hpp>` and to use the BOOST_TRY macros from the previous Carla Patches.
-- [rational.hpp.txt](https://github.com/carla-simulator/carla/files/8977465/rational.hpp.txt)
-- [read.hpp.txt](https://github.com/carla-simulator/carla/files/8977466/read.hpp.txt)
- Comment out building tests (sorry...) in LibCarla CMake file and BuildTools Setup.sh 
 --[CMakeLists.txt](https://github.com/carla-simulator/carla/files/8977506/CMakeLists.txt)
 --[Setup.sh.txt](https://github.com/carla-simulator/carla/files/8977508/Setup.sh.txt)
- Follow the rest of the normal installation instructions, however, build API with `make PythonAPI ARGS="--python-version=3.10"`

Hopefully, somebody can test it and use that for a PR maybe :)

dayaguec, 2022-09-15:
Hi everyone,

We are in the same circumstances wanting to use Carla 0.9.13 (latest release right now) with Ubuntu 22 and ROS 2 Humble. I can confirm that this workaround led to a successful compilation of Carla PythonAPI against Python 3.10. However, testing the generated egg file for my platform it seems that it is not always working. I tested this in two different set ups, Ubuntu 18 and Ubuntu 20, following the instructions posted here and using Python 3.10.4 installed from source and Python 3.10.5 installed from a deadsnakes PPA respectively. In both situations the egg file seems to work properly, using in Ubuntu 22 the precompile 0.9.13 release package of Carla.

However, when testing the generate_traffic.py script to generate vehicles and walkers in Carla world, the server crashes and the client produces a segmentation fault with no error or warning in the terminal only a report of Ubuntu with a “Python 3.10 crashed with a SIGSEGV (reading NULL VMA) in PyType_GenericAlloc()”. I debugged that particular script and found that the function get_random_location_from_navigation(), which retrieves a random location to be used as a destination using the go_to_location() method in carla.WalkerAIController is causing the segmentation fault. This means that other functions similar to this one may be affected in the same way having this setup.

Maybe this error is related with the changes applied to boost before compiling or maybe it is just a coherent error taking into consideration that we are using the client with a set up different from the one used by the original developers. I was also able to compile with both, boots 1.75 and the original boost 1.72 (it was not necessary to comment Gtests) and in both cases this error still appears. After looking at the code I realized  that this function is the only one affected by the CALL_RETURNING_OPTIONAL_WITHOUT_GIL in [libcarla.cpp](https://github.com/carla-simulator/carla/blob/a1b37f7f1cf34b0f6f77973c469926ea368d1507/PythonAPI/carla/source/libcarla/libcarla.cpp), maybe this is related.

I do not know if I am missing something but I hope this helps in a future workaround or maybe a new release for those using Carla in new Ubuntu versions.

WangwangZhu, 2023-03-28:
Hi, I am trying to compile the carla in Ubuntu 22.04 with ROS2 humble, I am stacking by the clang version, I want to ask which version of clang you used to while running " make PythonAPI"?, Thanks a lot.

dayaguec, 2023-03-28:
> Hi, I am trying to compile the carla in Ubuntu 22.04 with ROS2 humble, I am stacking by the clang version, I want to ask which version of clang you used to while running " make PythonAPI"?, Thanks a lot.

Taking into account my experience with this matter, I do not think Carla can be build in Ubuntu 22.04 right now since the compilation tool chain contains a lot of dependencies and incompatibilities for this Ubuntu distribution. Besides, Carla developers stated in the documentation that it is not supported for the time being. Having that said, what I did in my case was building Carla in Ubuntu 20.04 and compile the Python API against Python 3.10 to produce an eggfile for the binary version (since the binary version works on Ubuntu 22.04). Then you can use Carla with Python 3.10 and ROS 2 Humble that uses Python 3.10.

WangwangZhu, 2023-03-30:
> Taking into account my experience with this matter, I do not think Carla can be build in Ubuntu 22.04 right now since the compilation tool chain contains a lot of dependencies and incompatibilities [...]

Thanks for your suggestion~ I have followed your suggestion, and compiled the custommed map in Ubuntu 20.04 and copy the package to Ununtu 22.04. Carla worked well now. 
While I am compile the PythonAPI, the command "make PythonAPI" failed, the ERROR showed that I using the wrong clang version. I want to ask what's the version you used while building the PythonAPI with python 3.10

dayaguec, 2023-03-30:
> > > Hi, I am trying to compile the carla in Ubuntu 22.04 with ROS2 humble, I am stacking by the clang version, I want to ask which version of clang you used to while running " make PythonAPI"?, Thanks a lot.
> > 
> > 
> > Taking into account my experience with this matter, I do not think Carla can be build in Ubuntu 22.04 right now since the compilation tool chain contains a lot of dependencies and incompatibilities for this Ubuntu distribution. Besides, Carla developers stated in the documentation that it is not supported for the time being. Having that said, what I did in my case was building Carla in Ubuntu 20.04 and compile the Python API against Python 3.10 to produce an eggfile for the binary version (since the binary version works on Ubuntu 22.04). Then you can use Carla with Python 3.10 and ROS 2 Humble that uses Python 3.10.
> 
> Thanks for your suggestion~ I have followed your suggestion, and compiled the custommed map in Ubuntu 20.04 and copy the package to Ununtu 22.04. Carla worked well now. While I am compileingthe PythonAPI, the command "make PythonAPI" failed, the ERROR showed that I using the wrong clang version. I want to ask what's the version you used while building the PythonAPI with python 3.10

Oh, ok. Following the build instructions for [linux](https://carla.readthedocs.io/en/0.9.14/build_linux/), Ubuntu 20.04 I used: clang-10 lld-10 and g++-7. Remember to update the alternatives when building:

`sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-10/bin/clang++ 180 &&
sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-10/bin/clang 180`

WangwangZhu, 2023-04-01:
> Oh, ok. Following the build instructions for [linux](https://carla.readthedocs.io/en/0.9.14/build_linux/), Ubuntu 20.04 I used: clang-10 lld-10 and g++-7. Remember to update the alternatives when [...]

While installing clang-10, did you encouter this problems:
E: Package 'clang-10' has no installation candidate
E: Unable to locate package lld-10
E: Package 'g++-7' has no installation candidate
E: Package 'python' has no installation candidate
E: Package 'python-dev' has no installation candidate

dayaguec, 2023-04-01:
> While installing clang-10, did you encouter this problems: E: Package 'clang-10' has no installation candidate E: Unable to locate package lld-10 E: Package 'g++-7' has no installation candidate E: [...]

Add apt sources:
`sudo apt-add-repository "deb http://apt.llvm.org/focal/ llvm-toolchain-focal main"`

Follow the build instructions from the [documentation](https://carla.readthedocs.io/en/0.9.14/build_linux/).

Andrey9491, 2023-08-16:
https://github.com/gezp/carla_ros

roshambo919, 2023-10-06:
We've incorporated these updates in our fork: https://github.com/avstack-lab/carla/tree/build_0.9.13_py310_ubuntu22

See how we're using this with other python 3.10 libraries here: https://github.com/avstack-lab/carla-sandbox

trand2k, 2023-11-15:
> https://github.com/gezp/carla_ros

good job

Blyron, 2026-11-12:
Implemented: the Python API builds and runs on Python 3.10 to 3.15 (CMake option `CARLA_PYTHON_VERSIONS`), and release packages ship a wheel per version. See [Quick start](https://github.com/carla-simulator/carla/blob/ue58-dev/Docs/start_quickstart.md). Closing.

closed 2026-11-12 by Blyron