carla-simulator/carla #5544: Add new lidar type on server side

KPicherit, 2022-06-28:
Hello Carla team ! 

I'm having a hard time adding a new lidar type !

**CARLA version** : 0.9.13
**Platform/OS** : Ubuntu20.04
**Problem you have experienced**: I'm using a lidar that doesn't rotate around Z axis, hence cannot be represented through existing lidar types (RayCast/SemanticRayCast). I'd like to add a new lidar to the available sensors. 
I thought I'd reuse existing semantic lidar sensor two tweak it to my need, so I duplicated the SemanticRayCast lidar. First, before any tweaking, I'd like the new sensor to be available "as is" (i.e. with default rotating properties), so :
* In Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/
  * Duplicated RayCastSemanticLidar.cpp & .h, named it MyLidar.cpp /.h
  * Renamed the class from ARayCastSemanticLidar to AMyLidar
  * In AMyLidar::::GetSensorDefinition() method, changed TEXT("ray_cast_semantic") to TEXT("my_lidar")
* In /home/barth/Projects/carla/karla1/LibCarla/source/carla/sensor
  * Added the sensor to SensorRegistry.h as detailed within the .h file. 
    * Since the data format my lidar will be generating will look alike the SemanticLidar one, I used : std::pair<AMyLidar *, s11n::SemanticLidarSerializer>
  * Note : I also tried to duplicate the serializer (s11n) + data files (SemanticLidarData.h + SemanticLidarMeasurement.h) for MyLidar to have its own Serializer etc... in case it would help, but haven't been more successfull this way. (Note: in this case I also updated PythonAPI/carla/source/libcarla/SensorData.cpp so it knows about "MyLidar" measurements, detection and so on)
* In Plugins/Carla/Source/Carla/Actor/ActorBlueprintFunctionLibrary.cpp
  * Added an else/if at the end of UActorBlueprintFunctionLibrary::MakeLidarDefinition(..) to take into account my lidar : `else if (Id == "my_lidar") {..}` 

In the end, everything compiles when I duplicate the Serializer, Data, Measurement, and I can run the editor, but as soon as I use the PythonAPI (launch vehicle_gallery.py for instance) I get a core dump on the python client side.

What you expected to happen: Be able to spawn and retrieve data from a custom lidar sensor.
Steps to reproduce: Already explained above
Other information (documentation you consulted, workarounds you tried):
* Followed tutorial for the `Safe Distance Sensor`  at https://carla.readthedocs.io/en/latest/tuto_D_create_sensor/#appendix-client-side-sensors : it worked 
* Didn't find anything related to developing lidar sensors on server side in the issues nor in the documentation

**Side question** : is their something preventing from having two sensors using the same data and serializer ? (i.e. something like     ```std::pair<ARayCastSemanticLidar *, s11n::SemanticLidarSerializer>,
std::pair<AMyLidar *, s11n::SemanticLidarSerializer>, ```
in SensorRegistry.h

Any help on this would be much appreciated !

hashim19, 2022-08-05:
Hi, I am facing the exact same issue. I am trying to create a livox lidar in Carla. At first, I created this new lidar but couldn't get any success but then I just copied everything from RaycastLidar sensor and changed the names as you did. I have the exact same error as you do "Segmentation fault (core dumped)". I think it is something to do with sending data. Were you able to figure this out?

hashim19, 2022-08-08:
So, it looks like having the same data and serializer is an issue. But that is not the only problem. I did the following tests, 

1. I created a new sensor by copying everything from RayCastLidar sensor, however, the data and serializer were the same. In this situation, I was continuously getting this error "Segmentation fault (core dumped)". 
2. After that, I removed the the RayCastLidar.h and RayCastLidar.cpp files and removed the corresponding entries in the SensorRegistry.h file. By just doing that, the newly added lidar started working. 
3. So, I created the data and serializer with the new class and file names for my new lidar sensor. I brought back the RayCastLidar.h and RayCastLidar.cpp files. Also uncommented the RayCastLidar configuration in the SensorRegistry.h  This time, I got the "Segmentation fault (core dumped)" error again. 
4. I removed the RayCastLidar.h and RayCastLidar.cpp file again and commented the corresponding configuration in the SensorRegistry.h. The new lidar sensor started working with the newly added serializer and data classes and files. 

So now, I am not sure what the issue is. Why is having RayCastLidar and LivoxLidar files at the same time is causing "Segmentation fault (core dumped)" error. Let me know if you are able to find something.

cary1000h, 2022-11-24:
Hey guys, any updates on such adding new lidar routine? I was about to do the duplication trick as well and having lots of issues in rebuilding the pythonAPI. So I was wondering if this could work or not.

KPicherit, 2022-11-24:
Sorry, I have been working on other things lately. Hope I'll be able to come back to it. I'll post the progresses here if any 🤞

KPicherit, 2023-03-03:
Hello guys,

I Finally managed to make it work 🥳.
Anyone still interested by this ?

Satyam-53, 2023-10-23:
> Hello guys,
> 
> I Finally managed to make it work 🥳. Anyone still interested by this ?

Hi, yes. 
How did you make it work?

chrisgio91, 2023-10-23:
> Hello guys,
> 
> I Finally managed to make it work 🥳. Anyone still interested by this ?

Hi, I am interested too

Blyron, 2024-04-12:
Hello!

You can check logs of that segmentation fault at CARLA_ROOT/ Unreal/CarlaUE4/Saved/Logs 
Check those logs and it may give you a hint of what is going on. Without logs I cannot help you much.

Could you supply me with those? 
We will try to create a how to create a new LIDAR sensor

Blyron, 2024-04-19:
I am closing the issue feel free to reopen

closed 2024-04-19 by Blyron