carla-simulator/carla #5546: Adding new vehicle to CARLA with control model different from car

golnazraja, 2022-06-29:
Hello CARLA team,
I want to see if it's possible to add mobile robot models like rover, heavy machines, or turtle robot to CARLA and how. According to documentation, only car models are supported.
General case would be adding a model with Unified Robotics Description Format (URDF) format and being able to control it.

I hope you can help me with this issue since I searched a lot to find a solution for it.

fangchaooo, 2022-07-15:
em..... It is possible you need a lot of work:
1. create model animation
2. create control blueprint
3. create carla api
4. etc...

MahirGulzar, 2022-11-20:
> 

Is there any tutorial for this?

PeithonKing, 2023-01-12:
@fangchaooo can you please elaborate?
Also, could you tell me if I should use Carla for these particular cases, where we are not working with something as simple as a car (only moving parts are 4 wheels) instead we have a complex rover to work with. If your answer is somewhat like "for robotics there are better simulation engines than carla" can you suggest me any, please?

I want to do Reinforcement Learning on a rover model (I have an URDF file or an SLDASM file).