carla-simulator/carla #5551: Moving object labels

strohhut22, 2022-06-30:
Hello @lshafir.

> but we saw that the current buffers available to mark the object ID are only 0-255

That's our problem too!
To make clear, you are interested **only** in knowing if the object is moving or not?
The way I would do it would be checking the car transformation, and see if it is different from the previous frame. If this is the case, you can iterate all the points in your point cloud relevant to this car and set the information (like you said). How to find these points? Check out this issue in case you don't have done it #76. This is all we know on this topic.
We have plans to do this instance semantic segmentation in the future.

_Originally posted by @marcgpuig in https://github.com/carla-simulator/carla/issues/755#issuecomment-418405820_

Thanks for your great work on the Carla simulator and the instance segmentation that has been implemented is awesome!

I am having a question about moving objects: Is there a straightforward way to get annotations of moving or static actors in the current version of Carla (0.9.13)? 

Right now, given the instance segmentation, I am comparing the instance centers of objects to the previous frame. While this approach works like a charm, some minor false positives occur when points appear or disappear from a stationary object due to, e.g., occlusion. That is, because the bounding box center or mean of an instance's points change.

The only other way I see is implementing a new kind of sensor that adds this information to the semantic lidar sensor, is this the only way?

strohhut22, 2022-06-30:
If anyone is interested in a solution: Using all the transforms synchronized at world tick works combined with the instance labels.

closed 2022-06-30 by strohhut22

Kin-Zhang, 2023-02-13:
But I think there is still a problem since the instance labels are different between UE4 and CARLA, which causes the id you get from API is UE4 but the speed look up is CARLA actor.

Check this discussion maybe help: https://github.com/carla-simulator/carla/discussions/5047